riscv-software-src/riscv-pk #7: error: Cannot find simulator for target

santhoshvlsi, 2015-01-12:
While run the RISCV-PK we are getting the errors like 
Command: ../configure --prefix=/*PATH*/RISCV/riscv64-unknown-elf --host=riscv64-unknown-elf 

checking build system type... x86_64-unknown-linux-gnu
checking host system type... riscv64-unknown-elf
checking for riscv64-unknown-elf-gcc... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for riscv64-unknown-elf-g++... no
checking for riscv64-unknown-elf-c++... no
checking for riscv64-unknown-elf-gpp... no
checking for riscv64-unknown-elf-aCC... no
checking for riscv64-unknown-elf-CC... no
checking for riscv64-unknown-elf-cxx... no
checking for riscv64-unknown-elf-cc++... no
checking for riscv64-unknown-elf-cl.exe... no
checking for riscv64-unknown-elf-FCC... no
checking for riscv64-unknown-elf-KCC... no
checking for riscv64-unknown-elf-RCC... no
checking for riscv64-unknown-elf-xlC_r... no
checking for riscv64-unknown-elf-xlC... no
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for riscv64-unknown-elf-ar... no
checking for ar... ar
checking for riscv64-unknown-elf-ranlib... no
checking for ranlib... ranlib
checking for a BSD-compatible install... /usr/bin/install -c
checking for riscv64-unknown-elf-spike... no
checking for spike... no
configure: error: Cannot find simulator for target

sbeamer, 2015-01-12:
Can you verify your install of riscv64-unknown-elf-gcc worked?

santhoshvlsi, 2015-01-13:
Hi sbeamer,

How can i verify the riscv64-unknown-elf-gcc worked or not?
In installation bin path we have 
riscv64-unknown-elf-gcc         riscv64-unknown-elf-gprof    riscv64-unknown-elf-ranlib   spike-dasm
fesvr-eth  riscv64-unknown-elf-c++      riscv64-unknown-elf-gcc-4.9.2   riscv64-unknown-elf-ld       riscv64-unknown-elf-readelf  termios-xspike fesvr-rs232     riscv64-unknown-elf-c++filt  riscv64-unknown-elf-gcc-ar      riscv64-unknown-elf-ld.bfd   riscv64-unknown-elf-size     xspike
fesvr-zedboard                 riscv64-unknown-elf-cpp      riscv64-unknown-elf-gcc-nm      riscv64-unknown-elf-nm       riscv64-unknown-elf-strings riscv64-unknown-elf-addr2line  riscv64-unknown-elf-elfedit  riscv64-unknown-elf-gcc-ranlib  riscv64-unknown-elf-objcopy  riscv64-unknown-elf-strip
riscv64-unknown-elf-ar         riscv64-unknown-elf-g++      riscv64-unknown-elf-gcov        riscv64-unknown-elf-objdump  spike.

sbeamer, 2015-01-13:
Write a simple hello world program and attempt to compile it with riscv64-unknown-elf-gcc. If the compiler errors out, you will know you need to redo your riscv-gcc install.

Based on the info you posted, it appears autoconf doesn't have access to the same PATH.

santhoshvlsi, 2015-01-14:
Hi,
 Actually the issue is with installing RISCV PK.  After downloading the dump, i used the following  Commamd,  
  CC=riscv64-unknown-elf-gcc ../configure --prefix=$path/riscv64-unknown-elf --host=riscv64-unknown-elf
but it says  C compiler cannot create executables..!!!   How can i solve this problem..??

checking build system type... x86_64-unknown-linux-gnu
checking host system type... riscv64-unknown-elf
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc
checking whether the C compiler works... no
configure: error: in /Downloads/RISCV/Toolchain/riscv-pk/build':
configure: error: C compiler cannot create executables

sbeamer, 2015-01-14:
Yes, I understand the concern is over riscv-pk. Because autoconf complains the C compiler can't create executables (it was able to find riscv64-unknown-elf-gcc), I recommend you first verify that riscv64-unknown-elf-gcc was installed correctly. It is possible that the build process can complete, but the resulting binaries will not run (often due to a linking issue). I recommend attempting to compile a hello world program rather than running the compiler with no arguments to test if it is installed correctly.

santhoshvlsi, 2015-01-20:
1. Compile the Hello world program and we are getting the executable file HHH
riscv64-unknown-elf-gcc -o HHH hello.c.
2. Generated the Dump file.The command is 
riscv64-unknown-elf-objdump --disassemble-all HHH > d_hello.dump.
3. Generated the Hexadecimal file. 
elf2hex 16 16384 HHH > elfhex_hello. 

While simulate the HHH file in spike simulator(2 ways)
1. "spike pk HHH"
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open pk

2.  "spike HHH"  or "spike -d HHH" 
then we are geting the display in terminal
core   0: exception trap_illegal_instruction, epc 0x0000000000000000
core   0: 0x0000000000000000 (0x00001000) unknown
core   0: exception trap_illegal_instruction, epc 0x0000000000000000

How can i solve this problem..??

sbeamer, 2015-01-20:
Since you are able to get HHH, it appears your riscv64-unknown-elf-gcc may be working. I would try building riscv-pk again and examining config.log if autoconf fails again. Your trace from above had the error "checking whether the C compiler works... no" which means autoconf was able to find riscv64-unknown-elf-gcc but it did not pass autoconf's sanity checks. Hopefully config.log will contain more information you can use to debug your installation.

santhoshvlsi, 2015-01-22:
Run these command  CC=riscv64-unknown-elf-gcc ../configure --prefix=/home/riscv_installation/riscv64-unknown-elf-gcc --host=riscv After theat In bulid folder we are geting the config.log file.

This is the log file.

This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.
It was created by RISC-V Proxy Kernel configure ?, which was
generated by GNU Autoconf 2.69.  Invocation command line was
  $ ../configure --prefix=/home/riscv_installation/riscv64-unknown-elf-gcc --host=riscv
## --------- ##
## Platform. ##
## --------- ##
hostname = A-GA-78LMT-S2
uname -m = x86_64
uname -r = 3.13.0-44-generic
uname -s = Linux
uname -v = #73-Ubuntu SMP Tue Dec 16 00:22:43 UTC 2014
/usr/bin/uname -p = unknown
/bin/uname -X     = unknown
/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /usr/games
PATH: /usr/local/games
## ----------- ##
## Core tests. ##
## ----------- ##
configure:1941: checking build system type
configure:1955: result: x86_64-unknown-linux-gnu
configure:1975: checking host system type
configure:1988: result: mipsriscvel-ucb-elf
configure:2021: checking for riscv-gcc
configure:2048: result: riscv64-unknown-elf-gcc
configure:2317: checking for C compiler version
configure:2326: riscv64-unknown-elf-gcc --version >&5
riscv64-unknown-elf-gcc (GCC) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

configure:2337: $? = 0
configure:2326: riscv64-unknown-elf-gcc -v >&5
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
Target: riscv64-unknown-elf
Configured with: /home/Downloads/RISCV/Tchain/riscv-gnu-toolchain/src/newlib-gcc/configure --target=riscv64-unknown-elf --prefix=/home/riscv_installation --disable-shared --disable-threads --enable-tls --enable-languages=c,c++ --with-newlib --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls
Thread model: single
gcc version 4.9.2 (GCC) 
configure:2337: $? = 0
configure:2326: riscv64-unknown-elf-gcc -V >&5
riscv64-unknown-elf-gcc: error: unrecognized command line option '-V'
riscv64-unknown-elf-gcc: fatal error: no input files
compilation terminated.
configure:2337: $? = 1
configure:2326: riscv64-unknown-elf-gcc -qversion >&5
riscv64-unknown-elf-gcc: error: unrecognized command line option '-qversion'
riscv64-unknown-elf-gcc: fatal error: no input files
compilation terminated.
configure:2337: $? = 1
configure:2357: checking whether the C compiler works
[... 123 lines cut ...]
mandir='${datarootdir}/man'
oldincludedir='/usr/include'
pdfdir='${docdir}'
prefix='/home/riscv_installation/riscv64-unknown-elf-gcc'
program_transform_name='s,x,x,'
psdir='${docdir}'
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
stow=''
subprojects=''
subprojects_enabled=''
sysconfdir='${prefix}/etc'
target_alias=''
## ----------- ##
## confdefs.h. ##
## ----------- ##
/* confdefs.h */
#define PACKAGE_NAME "RISC-V Proxy Kernel"
#define PACKAGE_TARNAME "riscv-pk"
#define PACKAGE_VERSION "?"
#define PACKAGE_STRING "RISC-V Proxy Kernel ?"
#define PACKAGE_BUGREPORT "Andrew Waterman"
#define PACKAGE_URL ""

configure: exit 77

sbeamer, 2015-01-22:
I would verify that your riscv64-unknown-elf-gcc can compile a hello world and you get a binary out.

Looking at the log you posted, riscv64-unknown-elf-gcc can't find cc1. Is it possible you moved your riscv compiler after building it? It is best to build it with its final target directory given by --prefix.

PedroLSF, 2023-10-12:
I have the same problem, has anyone solved it?