riscv-software-src/riscv-pk #8: C compiler cannot create executables

santhoshvlsi, 2015-01-12:
Commamd: CC=riscv64-unknown-elf-gcc ../configure --prefix=/home/emupro/Documents/RISCV/riscv64-unknown-elf --host=riscv64-unknown-elf 

checking build system type... x86_64-unknown-linux-gnu
checking host system type... riscv64-unknown-elf
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc
checking whether the C compiler works... no
configure: error: in `/home/emupro/Downloads/RISCV/Toolchain/riscv-pk/build':
configure: error: C compiler cannot create executables

sbeamer, 2015-01-22:
This should be handled by the eventual resolution of issue #7. Please reopen this if it is not the case.

closed 2015-01-22 by sbeamer

mysoreanoop, 2020-08-11:
Running the following command (as per the README)
`../configure --prefix=$RISCV --host=riscv64-unknown-elf`
throws the following error (which is the same as posted by the OP):
```
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-elf
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc
checking whether the C compiler works... no
configure: error: in `/home/anoop/work/riscv-pk/build':
configure: error: C compiler cannot create executables
See `config.log' for more details
```
I have attached the config.log file that was generated in the build directory
[config.log](https://github.com/riscv/riscv-pk/files/5055854/config.log)

RISCV is set, and riscv-gnu-toolchain installed and appended to the PATH

mysoreanoop, 2020-08-11:
Sorry, the issue is resolved. 

I think the problem was that instead of riscv64-unknown-elf-gcc, the compilation process invoked riscv64-unknown-linux-gnu-gcc.

LakshmiKanth-77, 2021-03-30:
iam not able to interpret the solution. please help me

PedroLSF, 2023-10-12:
I need help too