riscv-software-src/riscv-pk #18: Cannot run an up-to-date Linux kernel in Spike or Qemu

manuelafm, 2016-03-23:
Hi,

(this problem should probably be tagged "help" or something, cannot find a way to achieve this before submitting).

I have problems running the examples below, it was working with a version from November.  I guess that this is because I am doing something wrong, or because the code is in flux in preparation for the next supervisor spec version.

I updated the repo and recompiled without making a copy of the binary before, so I don't have a working set-up now for my tests.  I tried to go back to the repo state by the start of November and several other points between then and now, but there are always errors compiling or it doesn't work (hangs without doing anything).  If you can tell me a way to get this working, perhaps by going back to a different repo commit, it would be very helpful.

Now onto the problem... When I compile with this line:
```
../configure --prefix=/usr/local/riscv64-unknown-linux-gnu --host=riscv64-unknown-linux-gnu
```

and then run with Spike, it complains:
```
$ spike +disk=root.bin bbl vmlinux
[logo]
       INSTRUCTION SETS WANT TO BE FREE
This is bbl's dummy_payload.  To boot a real kernel, reconfigure
bbl with the flag --with-payload=PATH, then rebuild bbl.
```

When I compile with the command line below (what I think to be the correct option for payload, I didn't find any concrete example), with a vmlinux image compiled on the same day:
```
../configure --prefix=/usr/local/riscv64-unknown-linux-gnu --host=riscv64-unknown-linux-gnu --with-payload=/archive/qemu-images/vmlinux
```
... then Spike just starts and hangs after the logo and "motto".

When I go back to mid November (and several other dates around that time), I always have this compilation problem:
```
make
riscv64-unknown-linux-gnu-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -I. -I../pk -I../softfloat -c ../pk/mtrap.c
riscv64-unknown-linux-gnu-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -I. -I../pk -I../softfloat -c ../pk/minit.c
../pk/mtrap.h: Assembler messages:
../pk/mtrap.h:208: Error: Instruction csrr requires absolute expression
../pk/mtrap.h:208: Error: Instruction csrr requires absolute expression
Makefile:299: recipe for target 'minit.o' failed
```

Cheers.

seldridge, 2016-04-05:
FWIW, I'm seeing the same problem after updating. Compiling with the `--with-payload=[full path to vmlinux]` produces a hang. Running this through spike with logging, I'm seeing it get stuck in the memcpy function called by `load_kernel_elf`. I'll do some more digging...

However, I'm somewhat curious as to the motivation for the BBL_PAYLOAD modifications. Does this mean that a bbl binary is tied to a specific kernel?

seldridge, 2016-04-05:
I was a little premature with this. I think that memcpy just takes a long time if you're running spike with logging. It obviously gets past this because the logo gets printed. Still digging...

handong32, 2016-04-05:
I'm hitting the same issue, on spike it is repeatedly outputting this after printing the logo and motto:

core   0: exception trap_instruction_access_fault, epc 0x0000000000000000

zhemao, 2016-04-06:
Hi guys. We are currently working on moving our software over to the new privilege 1.9 ISA. As a result, the versions of some things might be out of sync with each other. 

If you want to use the stable 1.7 ISA, make sure you use the correct versions. You should use the master branch of riscv-tools (*not* master branch of riscv-pk) and build the version of bbl and spike provided as submodules. In addition, you should use master branch or linux-4.1.y-riscv branch of riscv-linux. These versions will allow you to use the old command line `spike +disk=root.bin bbl vmlinux`.

If you want to use the new 1.9 ISA, you will need to check out the priv-1.9 branch of riscv-tools and build spike and bbl from that branch. In addition, you will need to use priv-1.9 or linux-4.1.y-riscv-priv-1.9 branch of riscv-linux. As you've already discovered, this version does not include HTIF support in bbl or vmlinux. As a result, you will need to configure bbl with the --with-payload option. Another consequence of removing HTIF is that the kernel will not have block device support so "+disk" will not work. For now, you can get around this by booting from an initramfs ramdisk instead of from a disk image.

We are sorry for the inconvenience this has caused and the lack of communication from our end. We hope to have the privilege 1.9 ISA finalized and the software in a stable state soon.

closed 2016-04-06 by zhemao

seldridge, 2016-04-06:
All good. I've gotten myself into trouble here because I've assumed that rocket-chip master was somewhat stable, but that's been aggressively tracking the 1.9 WIP updates as of late.

Just to clarify, the `--with-payload` is for specifying the kernel, i.e., vmlinux?

Also, you wouldn't happen to have an example for booting from an initramfs lying around? I'm out of my element here, but it seems like I'd need to configure the kernel to use an initramfs, package up the existing root filesystem image as an initramfs, and give this to the kernel at configure time so that it will automatically pull it in?

seldridge, 2016-04-06:
I was able to get through what I think is a correct initramfs setup within the kernel just using menuconfig to enable initramfs and point it at a directory containing the root filesystem with a /init defined.

Using `--with-payload` to point at this built vmlinux, I'm still seeing the behavior that @handong32 identified above, i.e., that core 0 exception trap/access fault on epc 0x0 repeating indefinitely.

Anyway, ignoring the result, does the methodology here seem correct?

seldridge, 2016-04-06:
*** I was on the wrong version of riscv-linux. I'm booting with spike and kernel panicking (progress!). This appears to be a problem with my /init. Yes, `--with-payload` does need to point at vmlinux.

Thanks for the help @zhemao -- your initial suggestion was enough to get me going here.