riscv-software-src/riscv-pk #20: spike pk test

susi655, 2016-05-11:
spike pk hello
warning: only got 2422210560 bytes of target mem (wanted 4294967296)
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open pk

How to solve this issue

gkelly, 2016-05-11:
That warning occurs when spike is trying to allocate memory but has to shrink it because the allocation fails.  However, I think that the actual error is due to spike not being able to find pk.  Is pk in the current directory?

susi655, 2016-05-12:
no

ccelio, 2016-05-12:
The `pk` is suppose to be in a particular location relative to Spike:

$RISCV/riscv64-unknown-elf/bin

You can manually specify the full path to your `pk`, and see if that works.

susi655, 2016-05-13:
spike: symbol lookup error: //usr/local/lib/libriscv.so: undefined symbol: _ZN6htif_t6mem_mbEv
I am unable to identify this error

gkelly, 2016-05-13:
That's the linker being unable to find `htif_t::mem_mb()`, which is from `libfesvr.so`.  Can you run `ldd spike` and see if it's resolving `libfesvr.so` properly?

susi655, 2016-05-13:
I added like
" ldd spike pk test"
Now it is giving

spike:
ldd: ./spike no such file or directory

pk:
ldd: ./pl no such file or directory

susi655, 2016-05-15:
The problem of symbol lookup error is coming when I type " spike pk hello"
the exact error is,
spike: symbol lookup error: //usr/local/lib/libriscv.so: undefined symbol: _ZN6htif_t6mem_mbEv
I am unable to identify this error

Can you please  tell the solution to this problem. Previously you told to add ldd before spike, but it is not giving solution. Can you please elaborate on this.

gkelly, 2016-05-15:
Okay, can you run:

    ldd `which spike`

And paste the output?

susi655, 2016-05-15:
ldd: ./which spike: No such file or directory

susi655, 2016-05-15:
Sorry!!!!
output is ...
	linux-vdso.so.1 =>  (0x00007ffea1dfc000)
	libriscv.so => //usr/local/lib/libriscv.so (0x00007f459db56000)
	libfesvr.so => //usr/local/lib/libfesvr.so (0x00007f459d92b000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f459d707000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f459d3fa000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f459d1e4000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f459ce1e000)
	libsoftfloat.so => //usr/local/lib/libsoftfloat.so (0x00007f459cc14000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f459c9f6000)
	/lib64/ld-linux-x86-64.so.2 (0x000055a480515000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f459c6ef000)

susi655, 2016-05-15:
output is ...
linux-vdso.so.1 => (0x00007ffea1dfc000)
libriscv.so => //usr/local/lib/libriscv.so (0x00007f459db56000)
libfesvr.so => //usr/local/lib/libfesvr.so (0x00007f459d92b000)
libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f459d707000)
libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f459d3fa000)
libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f459d1e4000)
libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f459ce1e000)
libsoftfloat.so => //usr/local/lib/libsoftfloat.so (0x00007f459cc14000)
libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f459c9f6000)
/lib64/ld-linux-x86-64.so.2 (0x000055a480515000)
libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f459c6ef000)

gkelly, 2016-05-15:
Okay, and now:

    nm -C /usr/local/lib/libfesvr.so | grep htif_t::mem_mb

susi655, 2016-05-15:
I did.
But same error for "spike pk hello"
i.e. spike: symbol lookup error: //usr/local/lib/libriscv.so: undefined
symbol: _ZN6htif_t6mem_mbEv

gkelly, 2016-05-15:
I need the output from:

    nm -C /usr/local/lib/libfesvr.so | grep htif_t::mem_mb

susi655, 2016-05-15:
no output.

gkelly, 2016-05-15:
Okay, that means that the symbol doesn't exist in that library.  I suspect that you might have an old copy of `libfesvr.so` installed.  Can you update your `riscv-fesvr` and `riscv-isa-sim` repositories, rebuild, reinstall and try again?

susi655, 2016-05-15:
i had taken from git repository only. Can you tell where the new ones are?

gkelly, 2016-05-15:
https://github.com/riscv/riscv-fesvr and https://github.com/riscv/riscv-isa-sim.  Build and install `riscv-fesvr` first, then `riscv-isa-sim`.  Make sure the `--prefix` you specify is the same for both.  Also, choose a new `--prefix` that you haven't used yet to ensure you're not mixing versions.

susi655, 2016-05-15:
I am following build.sh which was given in the github.

I have a general doubt? Just to know..
Why these tools are making problem with installation?
Why can't we have a standardized procedure, so that no one gets error?