riscv-software-src/riscv-pk #21: Regarding ABI

susi655, 2016-05-19:
When I ran  "riscv64-unknown-elf-gcc -m32 check.c"

It gave response..
/home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/bin/ld: /home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/lib/crt0.o: ABI is incompatible with that of the selected emulation
;
;
;
;
/home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/bin/ld: /home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/lib/crt0.o: file class ELFCLASS64 incompatible with ELFCLASS32
/home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: File in wrong format
collect2: error: ld returned 1 exit status

Is there any issue with version? How to solve this issue?

susi655, 2016-05-19:
When I ran "riscv64-unknown-elf-gcc -m32 check.c"

It gave response..
/home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/bin/ld: /home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/lib/crt0.o: ABI is incompatible with that of the selected emulation
;
;
;
;
/home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/bin/ld: /home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/lib/crt0.o: file class ELFCLASS64 incompatible with ELFCLASS32
/home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: File in wrong format
collect2: error: ld returned 1 exit status

Is there any issue with version? How to solve this issue?

palmer-dabbelt, 2016-05-19:
This is the wrong project to submit this bug to: "riscv64-unknown-elf-gcc" is GCC, which is part of riscv-gnu-toolchain.

You're probably getting this error because you don't have 32-bit libraries installed.  Try running configure in riscv-gnu-toolchain with "--with-xlen=32", which will build a 32-bit toolchain.

palmer-dabbelt, 2016-05-19:
Don't submit the same issue twice.

https://github.com/riscv/riscv-gnu-toolchain/issues/147

closed 2016-05-19 by palmer-dabbelt