riscv-software-src/riscv-pk #25: Compile error, "Instruction csrw requires absolute expression"

dbittman, 2016-07-05:
Compiling with the latest toolchain built by riscv-tools results in:

`riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"dummy_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../machine/minit.c
../machine/minit.c: Assembler messages:
../machine/minit.c:29: Error: Instruction csrw requires absolute expression
../machine/minit.c:30: Error: Instruction csrw requires absolute expression
../machine/mtrap.h:24: Error: Instruction csrr requires absolute expression
../machine/minit.c:47: Error: Instruction csrw requires absolute expression
../machine/minit.c:48: Error: Instruction csrw requires absolute expression
../machine/minit.c:49: Error: Instruction csrr requires absolute expression
../machine/minit.c:50: Error: Instruction csrr requires absolute expression
../machine/minit.c:150: Error: unrecognized opcode `mret'
make: *** [Makefile:300: minit.o] Error 1`

palmer-dabbelt, 2016-07-05:
This is usually just a version mismatch (what the error message actually means is that a CSR with the given name doesn't exist).  You should try using the matched tagged versions is riscv-tools, probably using whatever is on the "priv-1.9" branch if you're looking for new stuff.

dbittman, 2016-07-05:
Ah, I see. Solved, thanks.

closed 2016-07-05 by dbittman

barbara-gigerl, 2016-08-22:
I am still having the same problem....I get this when trying to make pk.
First I do: ../configure --prefix=/opt/riscv32i/riscv32-unknown-elf --host=riscv32-unknown-elf which seems ok and then when running make I get this:

> riscv32-unknown-elf-ranlib libsoftfloat.a
> riscv32-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"dummy_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../machine/mtrap.c
> riscv32-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"dummy_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../machine/minit.c
> ../machine/minit.c: Assembler messages:
> ../machine/minit.c:29: Error: Instruction csrw requires absolute expression
> ../machine/minit.c:30: Error: Instruction csrw requires absolute expression
> ../machine/mtrap.h:24: Error: Instruction csrr requires absolute expression
> ../machine/minit.c:47: Error: Instruction csrw requires absolute expression
> ../machine/minit.c:48: Error: Instruction csrw requires absolute expression
> ../machine/minit.c:49: Error: Instruction csrr requires absolute expression
> ../machine/minit.c:50: Error: Instruction csrr requires absolute expression
> ../machine/minit.c:159: Error: unrecognized opcode `mret'
> Makefile:300: recipe for target 'minit.o' failed
> make: *** [minit.o] Error 1

Edgar-Conzen, 2016-08-25:
Similar problem here as Barbara
First I tried to set up a 32-bit toolchain, following the read.me on https://github.com/riscv/riscv-tools
with a slightly modified build.sh :
...
build_project riscv-gnu-toolchain --prefix=$RISCV --with-xlen=32 --with-arch=IC
CC= CXX= build_project riscv-pk --enable-32bit --prefix=$RISCV/riscv32-unknown-elf --host=riscv32-unknown-elf

The riscv-gnu-toolchain built well but building riscv-pk failed with:
...
Configuring project riscv-pk
configure: WARNING: unrecognized options: --enable-32bit
configure: WARNING: using cross tools not prefixed with host triplet
configure: WARNING: unrecognized options: --enable-32bit
Building project riscv-pk
../pk/mentry.S: Assembler messages:
../pk/mentry.S:192: Error: unrecognized opcode `amoor.w x0,a1,(a2)'
make: *** [mentry.o] Error 1

...

So I did a compare with the newest revision of riscv-pk and found that --enable-32bit isn't
a valid argument in the older version that you get with https://github.com/riscv/riscv-tool

I then replaced the old version of riscv-pk with the newest one and got:

Configuring project riscv-pk
Building project riscv-pk
../machine/minit.c: Assembler messages:
../machine/minit.c:29: Error: Instruction csrw requires absolute expression
../machine/minit.c:30: Error: Instruction csrw requires absolute expression
../machine/mtrap.h:24: Error: Instruction csrr requires absolute expression
../machine/minit.c:47: Error: Instruction csrw requires absolute expression
../machine/minit.c:48: Error: Instruction csrw requires absolute expression
../machine/minit.c:49: Error: Instruction csrr requires absolute expression
../machine/minit.c:50: Error: Instruction csrr requires absolute expression
../machine/minit.c:159: Error: unrecognized opcode `mret'
make: *** [minit.o] Error 1

I did a google search that led me to this page but I am not able to determine
the 'matched tagged versions of riscv-tools' that palmer-dabbelt recommended

barbara-gigerl, 2016-08-26:
Me too, riscv-gnu-toolchain builds well but pk won't. I also tried an old version of pk and got the same result as Edgar! Can anyone please help?
I nearly tried everything...I modified configure a, I used different options but nothing helped.

drgprj, 2016-09-03:
Me four, I've tried following the steps of this Stack Overflow answer: http://stackoverflow.com/a/31393890/6779984. Tried with the "priv-1.9" branch on riscv-tools repository. 

I've tried running this:
`$ riscv64-unknown-elf-gcc -nostdlib -nostartfiles -T link.ld -o hello hello.s`

The following error shows up:
```
hello.s: Assembler messages:
hello.s:6: Error: Instruction csrw requires absolute expression
hello.s:6: Error: Instruction csrw requires absolute expression
```

ccelio, 2016-09-03:
If you are following the instructions in that stack overflow answer, they are woefully out-of-date for the latest v1.9 privileged architecture.  There is no tohost CSR register, so you can no longer end bare-metal tests using a `csrw tohost, 1`.  This is the current technique to end tethered tests, using memory mapped IO: https://github.com/riscv/riscv-test-env/blob/master/p/riscv_test.h#L119. 

I have since updated the SO answer with a disclaimer.

aditech123, 2018-08-10:
I am trying to compile the FreeRTOS and got the Same error

Error:
    CC ../../Source/portable/GCC/RISCV/port.c
../../Source/portable/GCC/RISCV/port.c: Assembler messages:
../../Source/portable/GCC/RISCV/port.c:121: Error: Instruction csrr requires absolute expression
../../Source/portable/GCC/RISCV/port.c:123: Error: Instruction csrw requires absolute expression
../../Source/portable/GCC/RISCV/port.c:130: Error: Instruction csrr requires absolute expression
../../Source/portable/GCC/RISCV/port.c:132: Error: Instruction csrw requires absolute expression
Makefile:123: recipe for target '../../Source/portable/GCC/RISCV/port.o' failed
make: *** [../../Source/portable/GCC/RISCV/port.o] Error 1

zetalog, 2023-07-30:
I've been directed here as I've been suffering with the same issue.
I should say sth here about what I've learned from solving this issue.

The issue might be caused by a wrong implementation of stringify used in inlined code.

For example:
This should work:
#define __stringify_1(x...)             #x
#define __stringify(x...)               __stringify_1(x)
While this shouldn't work in some inlined code:
#define __stringify(x...)               #x