riscv-software-src/riscv-pk #28: -mcmodel=medany problem when compiling

rain-1, 2016-07-20:
Hello! I have been trying to set up the RISC V tools, I got `riscv-gnu-toolchain` built and installed to my prefix but then when I try to build this `riscv-pk` I get an error:

```
$ make
gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"dummy_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/file.c
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
make: *** [Makefile:300: file.o] Error 1
```

Thanks for any input on solving this.

aswaterman, 2016-07-20:
Looks like it's using your host compiler, not the RISC-V target compiler.  Make sure you don't have the CC environment variable set, and that you set --host=riscv64-unknown-elf (or similar) when you configure pk.

rain-1, 2016-07-20:
Thank you very much!

closed 2016-07-20 by rain-1