riscv-software-src/riscv-pk #35: spike pk hello

susi655, 2016-08-31:
I installed 32-bit tool chain. (riscv32-elf-unknown-gcc)

When i check for spike pk hello, following error came..
$riscv32-unknown-elf-gcc -o hello hello.c
$spike pk hello
terminate called after throwing an instance of 'std::runtime_error'
what(): could not open pk

pk is in $RISCV/riscv32-unknown-elf/bin

alexmechanic, 2016-09-01:
same problem. also cannot build riscv-pk with errors:
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
when changed to 'medium' or 'kernel' getting an error:
../pk/file.c: Assembler messages:
../pk/file.c:28: Error: no such instruction: `fence'

palmer-dabbelt, 2016-09-01:
This sounds very much like

https://github.com/riscv/riscv-gnu-toolchain/issues/174

Can you check and see if my commit screwed it up?  It certainly smells like I screwed something up, because I actually have another open pull request that claims to do exactly the same thing, but didn't get merged.  Maybe whatever bumps Andrew claims were necessary in that PR just landed on priv-1.9 and didn't end up on master?

https://github.com/riscv/riscv-tools/pull/36

alexmechanic, 2016-09-02:
palmer-dabbelt
If this was addressed not to me (as i suppose the question is about 32-bit libs), im still confused about badly written Makefiles and sources

palmer-dabbelt, 2016-09-08:
@susi655 If you look at the PR I mentioned, it has a few commits that were necessary to make the 32-bit stuff actually work, but those commits were only on the priv-1.9 branches.  I've gone ahead and backported them to master, so everything should build now.  You can see everything on this branch

  https://github.com/riscv/riscv-tools/tree/tests-xlen

Unfortunately, my "Hello, World" test case doesn't print anything, so something else seems to be wrong somewhere with the 32-bit stuff.  Since it's so out of date as compared to the priv-1.9 stuff (which is where all the real work is going), I'm probably not going to have time to look at it for a while.

Qiyamah, 2019-03-25:
is anybody looking into this?
Thanks

Qiyamah, 2019-03-25:
```bash
 -fno-PIE -mabi=lp64d -march=rv64imafdc   -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -I. -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/pk -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/bbl -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/softfloat -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/dummy_payload -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/machine -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/util -c /home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/pk/elf.c
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
gcc: error: unrecognized argument in option ‘-mabi=lp64d’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
gcc: error: unrecognized argument in option ‘-mabi=lp64d’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE -mabi=lp64d -march=rv64imafdc   -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -I. -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/pk -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/bbl -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/softfloat -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/dummy_payload -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/machine -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/util -c /home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/pk/console.c
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
gcc: error: unrecognized argument in option ‘-mabi=lp64d’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE -mabi=lp64d -march=rv64imafdc   -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -I. -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/pk -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/bbl -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/softfloat -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/dummy_payload -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/machine -I/home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/util -c /home/rra/Projects/RISCV/freedom/freedom-u-sdk/riscv-pk/pk/mmap.c
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
Makefile:319: recipe for target 'file.o' failed
make[1]: *** [file.o] Error 1
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
gcc: error: unrecognized argument in option ‘-mabi=lp64d’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
make[1]: *** Waiting for unfinished jobs....
Makefile:319: recipe for target 'syscall.o' failed
make[1]: *** [syscall.o] Error 1
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
gcc: error: unrecognized argument in option ‘-mabi=lp64d’
Makefile:319: recipe for target 'handlers.o' failed
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
make[1]: *** [handlers.o] Error 1
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
gcc: error: unrecognized argument in option ‘-mabi=lp64d’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
Makefile:319: recipe for target 'frontend.o' failed
make[1]: *** [frontend.o] Error 1
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
Makefile:319: recipe for target 'elf.o' failed
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
make[1]: *** [elf.o] Error 1
gcc: error: unrecognized argument in option ‘-mabi=lp64d’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
Makefile:319: recipe for target 'console.o' failed
make[1]: *** [console.o] Error 1
Makefile:319: recipe for target 'mmap.o' failed
make[1]: *** [mmap.o] Error 1
make[1]: Leaving directory '/home/rra/Projects/RISCV/freedom/freedom-u-sdk/work/riscv-pk'
Makefile:175: recipe for target '/home/rra/Projects/RISCV/freedom/freedom-u-sdk/work/riscv-pk/bbl' failed
make: *** [/home/rra/Projects/RISCV/freedom/freedom-u-sdk/work/riscv-pk/bbl] Error 2

```

volglizolic, 2019-04-30:
in fedora install development tools and development libraries
sudo dnf groupinstall "Development Tools" "Development Libraries"

Debian based and ect have something similar

chyhsueh, 2019-07-22:
> same problem. also cannot build riscv-pk with errors:
> gcc: _error: unrecognized argument in option ‘-mcmodel=medany’_
> 
> when changed to 'medium' or 'kernel' getting an error:
> ../pk/file.c: Assembler messages:
> ../pk/file.c:28: _Error: no such instruction: `fence'_

Hi, buddies:

It seems the symptoms found in case the expected toolchain with prefix "riscv64-unknown-elf-" cannot be found in the host system. Currently the riscv-gnu-toolchain must be built/ready before the riscv-tools could be built. Please grab the source code of riscv-gnu-toolchain from the following repo: 

https://github.com/riscv/riscv-gnu-toolchain

The expected toolchain will be built by:

> ./configure --prefix=$RISCV
> make

Please following the section "Installation (Newlib)" in Readme.md. Once the riscv-gnu-toolchain was successfully setup in $RISCV and the directory $RISCV\bin was set in $PATH, the riscv-tools can be successfully built then.