riscv-software-src/riscv-pk #36: Makefile and sources have errors

alexmechanic, 2016-09-01:
I cannot build riscv-pk. error:
gcc: error: unrecognized argument in option ‘-mcmodel=medany’

when changed to 'medium' or 'kernel' getting an error:
../pk/file.c: Assembler messages:
../pk/file.c:28: Error: no such instruction: `fence'

alexmechanic, 2016-09-03:
for riscv32 works correctly, issue is only for 64-bit

aswaterman, 2016-09-03:
configure found an x86 compiler; make sure --host is set right and that the appropriate risc-v compiler is in your PATH

alexmechanic, 2016-09-05:
host was set as told in README. PATH is full

mwachs5, 2016-09-06:
I'm having the same issue. 

Do you mean that we need to set both RISCV and our PATH variable to RISCV:PATH before running the build scripts?

mwachs5, 2016-09-06:
I resolved this issue by compiling in a new shell with cleaner PATH. I think Xilinx tools were causing issues in my path.

aswaterman, 2016-09-06:
Oh... yeah... they have a different libstdc++ that isn't compatible with
newer software.

alexmechanic, 2016-09-07:
I don't have any other software can cause issues in PATH..

aswaterman, 2016-09-07:
Did you build both a 64b and 32b compiler?  If riscv64-unknown-elf-gcc isn't in your path, configure will get confused.

palmer-dabbelt, 2016-09-08:
There have been some commits to riscv-tools that should clean up some of these "configure finding an x86 compiler" problems.   Everything is on this branch

  https://github.com/riscv/riscv-tools/tree/tests-xlen

Can you check and see if that fixes your build issues?

alexmechanic, 2016-09-09:
build succeeds when building riscv-tools (build.sh) for both 32 and 64bit.

palmer-dabbelt, 2016-09-12:
It looks like the build.sh patches fixed things.

closed 2016-09-12 by palmer-dabbelt

mathmax12, 2019-08-05:
@alexmechanic 
Hi Alex,

I got the same issue as yours.
"riscv-pk/build$    ../configure --prefix=$RISCV --host=riscv64-unknown-elf
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-elf
checking for riscv64-unknown-elf-gcc... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for riscv64-unknown-elf-g++... no
checking for riscv64-unknown-elf-c++... no
checking for riscv64-unknown-elf-gpp... no
checking for riscv64-unknown-elf-aCC... no
checking for riscv64-unknown-elf-CC... no
checking for riscv64-unknown-elf-cxx... no
checking for riscv64-unknown-elf-cc++... no
checking for riscv64-unknown-elf-cl.exe... no
checking for riscv64-unknown-elf-FCC... no
checking for riscv64-unknown-elf-KCC... no
checking for riscv64-unknown-elf-RCC... no
checking for riscv64-unknown-elf-xlC_r... no
checking for riscv64-unknown-elf-xlC... no
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for riscv64-unknown-elf-ar... no
checking for ar... ar
checking for riscv64-unknown-elf-ranlib... no
checking for ranlib... ranlib
checking for riscv64-unknown-elf-readelf... no
checking for readelf... readelf
checking for riscv64-unknown-elf-objcopy... no
checking for objcopy... objcopy
checking for a BSD-compatible install... /usr/bin/install -c
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
configure: configuring default subproject : pk
configure: configuring default subproject : bbl
configure: configuring default subproject : softfloat
configure: configuring default subproject : dummy_payload
configure: configuring default subproject : machine
configure: configuring default subproject : util
configure: creating ./config.status
config.status: creating pk.mk
config.status: creating bbl.mk
config.status: creating softfloat.mk
config.status: creating dummy_payload.mk
config.status: creating machine.mk
config.status: creating util.mk
config.status: creating Makefile
config.status: creating config.h
config.status: config.h is unchanged
"
I tried to run the build.sh in the riscv-tools repo and then rerun the build but it doesn't work for me.
Is that what you did?

abdurrehmanashraf, 2023-11-01:
> build succeeds when building _riscv-tools_ repo with _build.sh_, for both 32 and 64bit.

hi how you did this with build.sh ?  i am getting error when install proxy kernal (pk) for spike [simam@localhost build]$ make
gcc -std=gnu11 -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=large -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE    -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/file.c
../pk/file.c: Assembler messages:
../pk/file.c:28: Error: no such instruction: `fence'
how you resolve this

abdurrehmanashraf, 2023-11-01:
> @alexmechanic Hi Alex,
>
> I got the same issue as yours. "riscv-pk/build$ ../configure --prefix=$RISCV --host=riscv64-unknown-elf checking build system type... x86_64-pc-linux-gnu checking host system [...]

download the source code and make directory in that i face same issue then notice no source code downloaded