riscv-software-src/riscv-pk #39: 32 bit skipe+pk

emelcher, 2016-09-21:
From what I understand from #35 , there is no clean way to compile code for RV32I and simulate it together with pk in spike, and this problem will not be fixed for a while.

I have found a workaround, however it lacks generality. Perhaps someone has ideas for improvement:

Given some code that one wants to simulate for RV32I in a file called lupi.c:
```
int lupi(register int x, register int y) {
   register int s;
   s = x+y;
   return s;
}
```
Use this command to compile it and obtain RV32I assembly code in file lupi.s:
```
riscv64-unknown-elf-gcc -m32 -march=RV32I -O1 -S lupi.c 
```
Create a file lupi_main.c that acts as interface between lupi and pk:
```
#include <stdio.h>
int lupi(int, int);
int main(int argc,char *argv[]) {
   int x,y,s;
   x = argv[1][0]-'0';       // 1st letter of 1st argument converted to a value between 0 e 9
   y = argv[2][0]-'0';       // 1st letter of 2nd argument
   s =  lupi(x,y);
   printf("Hello, result is %d.\n", s);
   return s;
}
```
Compile, assemble and link:
```
riscv64-unknown-elf-gcc lupi_main.c lupi.s -o lupi
```
Simulate:
```
spike pk lupi 7 2
```

emelcher, 2018-03-19:
I revisited this problem last week and found a way to compile the 32 bit toolchain from up to date code so there is no workaround needed any more.

However, you cannot use the `build.sh` from https://riscv.org/software-tools.
Here are the different configure options and changes to Makefiles to compile the toolchain:

riscv-gnu-toolchain:
`./configure --prefix=$RISCV --with-arch=rv32i --with-abi=ilp32 --with-xlen=32`

riscv-fesvr:
`../configure --prefix=$RISCV`
followed by:
`sed -i 's:CXXFLAGS      +=:CXXFLAGS      += -DTARGET_ARCH=\\"riscv32-unknown-elf\\":' Makefile`

riscv-pk:
`../configure --prefix=$RISCV  --host=riscv32-unknown-elf --enable-32bit`
followed by:
`sed -i 's:-m32::g' Makefile`

riscv-isa-sim:
`../configure --prefix=$RISCV --with-fesvr=$RISCV --with-isa=rv32i --enable-histogram --with-xlen=32`

The complete command sequence can be found at
http://labarc.ufcg.edu.br/loac/index.php?n=OAC.Masm#install

Now, one can use all commands normally:
`riscv32-unknown-elf-gcc -O1 -S lupi.c`
`riscv32-unknown-elf-gcc lupi_main.c lupi.s -o lupi`
`spike pk lupi 7 2`

The drawback is that bare metal spike (without pk) does not work in a simple way any more
https://stackoverflow.com/questions/31390127/how-can-i-compile-c-code-to-get-a-bare-metal-skeleton-of-a-minimal-risc-v-assemb
so I am glad to still have my toolchain from Sep 21, 2016 for this.

closed 2018-03-19 by emelcher

polypolen, 2018-05-07:
Thanks emelcher. 
@emelcher which versions of riscv-tools / riscv-gnu-toolchain / riscv-isa-sim are you using?

The configure option `--with-xlen` is not supported by riscv-gnu-toolchain:
```
$ ./configure --help | grep with-xlen
$ git log -1 master
commit 994cafdbe00b25fa3442efa163503b616fcfdcac
Merge: 1b80cbe a17086f
Author: Tim Newsome <tim@sifive.com>
Date:   Tue Apr 17 11:54:48 2018 -0700

    Merge pull request #332 from riscv/bump-20180413
    
    Bump gdb
```

nor by `riscv-isa-sim`:
```
$ git log -1
commit d48f107dba6a96fb827cb47fdf290261feadeb35
Author: Andrew Waterman <andrew@sifive.com>
Date:   Fri May 4 12:05:33 2018 -0700

    Revert "C.LWSP and C.LDSP with rd=0 are legal instructions"
    
    See https://github.com/riscv/riscv-isa-manual/commit/01190b6ebeb29cfac6783a3e7ce30cd529bf6c59
$ ./configure --help | grep with-xlen
$
```

emelcher, 2018-05-07:
I agree that `--with-xlen` probably has no effect at all, but I did not take the time to test this.

Here are the versions I used:
```
[riscv-gnu-toolchain]$ git log -1 master
commit 1b80cbe97d2c29212398d3b74ddc54718ad32e23
Merge: 7df3261 2d78afa
Author: Jim Wilson <jimw@sifive.com>
Date:   Wed Jan 31 17:51:30 2018 -0800

    Merge pull request #320 from riscv/expat-not-native
    
    Make linux-native builds work again.
[riscv-gnu-toolchain]$ cd ../riscv-fesvr
[riscv-fesvr]$ git log -1 master
commit 0d34fdbe104cbe0f07df6d9aceb2763f5fbba123
Author: Prashanth Mundkur <prashanth.mundkur@gmail.com>
Date:   Mon Mar 12 15:57:27 2018 -0700

    Decouple memif_t and htif_t. (#43)
    
    Define a chunked_memif_t interface that serves memif_t, and is implemented by htif_t and htif_hexwr
[riscv-fesvr]$ cd ../riscv-pk
[riscv-pk]$ git log -1 master
commit 2bbd8e1a1bccae13ec87882baf423abfc6ef76fd
Author: Wesley W. Terpstra <wesley@sifive.com>
Date:   Mon Mar 5 14:59:44 2018 -0800

    mtrap: add a halt IPI used for poweroff (#86)
    
    Otherwise, linux complains the moment an interrupt arrives and
    wakes up one of the not-looping cores.
[riscv-pk]$ cd ../riscv-isa-sim
[riscv-isa-sim]$ git log -1 master
commit 7e35a2a62f7433060e2ab1c98b3afd8b8a69b829
Author: Prashanth Mundkur <prashanth.mundkur@gmail.com>
Date:   Wed Mar 14 09:48:11 2018 -0700

    Fix a bug caused by moving misa into state_t. (#180)
    
    * Fix misa losing its value in processor constructor due to state:reset() following state.misa init
    Make state:reset() preserve misa.
    
    * Set state.misa to max_isa on reset().
    
    * Idiomatic fix for earlier commit.
```

polypolen, 2018-05-08:
Thanks @emelcher 

I wrote a shell script ---below--- that reproduces your build instructions, but I still experience the same problem.  It looks like `spike` does not find `pk` at its expected location. So I assume the problem is more on `spike`'s side.  Do you agree?

```
$ echo -e '#include <stdio.h>\n int main(void) { printf("Hello world!\\n"); return 0; }' > hello.c && riscv32-unknown-elf-gcc -o hello hello.c && spike pk hello
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open pk (did you misspell it? If VCS, did you forget +permissive/+permissive-off?)

$ echo -e '#include <stdio.h>\n int main(void) { printf("Hello world!\\n"); return 0; }' > hello.c && riscv32-unknown-elf-gcc -o hello hello.c && spike $RISCV/riscv32-unknown-elf/bin/pk hello
Hello world!
```

build script:
```
$ cat build-20180507.sh
#!/bin/bash

# https://github.com/riscv/riscv-pk/issues/39

[ -z "$TOP" ] && echo "TOP is not defined" && exit -1
[ -z "$RISCV" ] && echo "RISCV is not defined" && exit -1

cd $TOP
cd riscv-gnu-toolchain
git checkout 1b80cbe97d2c29212398d3b74ddc54718ad32e23
rm -rf build && mkdir build && cd build
../configure --prefix=$RISCV --with-arch=rv32i --with-abi=ilp32 --with-xlen=32
make -j5
make install

cd $TOP
cd riscv-fesvr
git checkout 0d34fdbe104cbe0f07df6d9aceb2763f5fbba123
rm -rf build && mkdir build && cd build
../configure --prefix=$RISCV
sed -i 's:CXXFLAGS +=:CXXFLAGS += -DTARGET_ARCH=\\"riscv32-unknown-elf\\":' Makefile
make
make install

cd $TOP
cd riscv-pk
git checkout 2bbd8e1a1bccae13ec87882baf423abfc6ef76fd
rm -rf build && mkdir build && cd build
../configure --prefix=$RISCV --host=riscv32-unknown-elf --enable-32bit
sed -i 's:-m32::g' Makefile
make
make install

cd $TOP
cd riscv-isa-sim
git checkout 7e35a2a62f7433060e2ab1c98b3afd8b8a69b829
rm -rf build && mkdir build && cd build
../configure --prefix=$RISCV --with-fesvr=$RISCV --with-isa=rv32i --enable-histogram --with-xlen=32
make
make install
```

emelcher, 2018-05-09:
I made some modifications to Your script, see below.
- added lines `git clone` in order to be able to start in am empty directory 
- deleted the  `make install` for riscv-gnu-toolchain
- changed the space character in the pattern of the first `sed` to match any number of whitespace characters

Now I get the correct result:
```
$ echo -e '#include <stdio.h>\n int main(void) { printf("Hello world!\\n"); return 0; }' > hello.c && riscv32-unknown-elf-gcc -o hello hello.c && spike pk hello
Hello world!
```

```
#!/bin/bash

# https://github.com/riscv/riscv-pk/issues/39

[ -z "$TOP" ] && echo "TOP is not defined" && exit -1
[ -z "$RISCV" ] && echo "RISCV is not defined" && exit -1

cd $TOP
git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git checkout 1b80cbe97d2c29212398d3b74ddc54718ad32e23
rm -rf build && mkdir build && cd build
../configure --prefix=$RISCV --with-arch=rv32i --with-abi=ilp32 --with-xlen=32
make

cd $TOP
git clone https://github.com/riscv/riscv-fesvr.git
cd riscv-fesvr
git checkout 0d34fdbe104cbe0f07df6d9aceb2763f5fbba123
rm -rf build && mkdir build && cd build
../configure --prefix=$RISCV
sed -i 's:CXXFLAGS\s\++=:CXXFLAGS += -DTARGET_ARCH=\\"riscv32-unknown-elf\\":' Makefile
make
make install

cd $TOP
git clone https://github.com/riscv/riscv-pk.git
cd riscv-pk
git checkout 2bbd8e1a1bccae13ec87882baf423abfc6ef76fd
rm -rf build && mkdir build && cd build
../configure --prefix=$RISCV --host=riscv32-unknown-elf --enable-32bit
sed -i 's:-m32::g' Makefile
make
make install

cd $TOP
git clone https://github.com/riscv/riscv-isa-sim.git
cd riscv-isa-sim
git checkout 7e35a2a62f7433060e2ab1c98b3afd8b8a69b829
rm -rf build && mkdir build && cd build
../configure --prefix=$RISCV --with-fesvr=$RISCV --with-isa=rv32i --enable-histogram --with-xlen=32
make
make install

```

emelcher, 2018-05-10:
I recompiled everything without `--with-xlen=32` and the hello-test still works.
Without all the `git checkout` lines however the hello-test seems to get caught in an endless loop.

polypolen, 2018-05-15:
On 10/05/2018 20:40, emelcher wrote:
> I recompiled everything without |--with-xlen=32| and the hello-test
> still works.
> Without all the |git checkout| lines however the hello-test seems to get
> caught in an endless loop.

thank you @emelcher.

So it looks like fesvr is missing a configuration option for 32bit
toolchains, in order to define TARGET_ARCH.

I have no time available for now, but I will try to propose a patch ASAP.

best,

-- 
Damien Couroussé
Research Engineer at CEA-LIST
DRT/LIST/DACLE/SCSN/LIALP
Laboratoire Infrastructures et Ateliers pour le Logiciel sur Puces
Commissariat à l’énergie atomique et aux énergies alternatives

MINATEC Campus | 17 rue des Martyrs | 38054 Grenoble Cedex 9
Tel : +33 (0)4 38 78 04 66
damien.courousse@cea.fr

[Citation aléatoire]
A: Because it messes up the order in which people normally read text.
   See http://en.wikipedia.org/wiki/Posting_style
Q: Why is top-posting such a bad thing?

polypolen, 2018-06-06:
for reference: this issue is fixed by commit `1a1edf1f0b` in `riscv-fesvr`.
https://github.com/riscv/riscv-fesvr/pull/47

GiuseppeDiGuglielmo, 2018-06-22:
I am here with a similar problem. I would like to bare-metal simulate _really basic_ RV32I binaries. At the moment, I followed the instructions at 
https://stackoverflow.com/questions/31390127/how-can-i-compile-c-code-to-get-a-bare-metal-skeleton-of-a-minimal-risc-v-assemb

In my understanding the assembly code should look like
```
hello.s:
 .align 6
 .globl _start
 _start:
#
# MY RV32I CODE GOES HERE!
#
 csrwi mtohost, 1;
 1:
 j 1b
```
@emelcher as you said few posts above, I checked out an old version of the tools from https://github.com/riscv/riscv-tools from the last commit of Sep 21, 2016. But that does not work as well.

What I noticed is so far:

- the CSR `mtohost` is not there anymore (at least it does not compile with it)
- if I drop the `csrwi mtohost, 1;` instruction, and run spike on the binary I get the following error:
```
$ spike --isa=RV32I rv32ui-p-add                                                                                                                                                                                
terminate called after throwing an instance of 'trap_store_access_fault'
```

Do you have any suggestions?

emelcher, 2018-06-22:
For bare metal, I am using a still older version of the toolchain.
After running the script I posted here on May 9, I run
```
cd $TOP
wget -qO- http://labarc.ufcg.edu.br/loac/uploads/OAC/riscv-bm.tgz | tar xz 
cd bm
cd riscv-fesvr/build
../configure --prefix=$RISCV/bm
make
make install
cd ../../riscv-isa-sim/build
../configure --prefix=$RISCV/bm   --with-fesvr=$RISCV/bm
make
make install
ln -s $RISCV/bm/bin/spike $RISCV/bin/spike-bm
```
When I realized that bare metal was broken, I managed to find these old files in one of the PCs of the lab and saved them in riscv-bm.tgz. Unfortunately I am unable to find out the git version from them.

emelcher, 2018-06-26:
Examining my files again I found the following checkout versions that work for bare metal:
```
cd $TOP
mkdir bm && cd bm
git clone https://github.com/riscv/riscv-fesvr.git
cd riscv-fesvr
git checkout 0f34d7ad311f78455a674224225f5b3056efba1d
mkdir build && cd build
../configure --prefix=$RISCV/bm
make
make install
cd ../..
git clone https://github.com/riscv/riscv-isa-sim.git
cd riscv-isa-sim
git checkout 3bfc00ef2a1b1f0b0472a39a866261b00f67027e
mkdir build && cd build
../configure --prefix=$RISCV/bm   --with-fesvr=$RISCV/bm
make
make install
ln -s $RISCV/bm/bin/spike $RISCV/bin/spike-bm
cd $TOP
```

emelcher, 2018-07-06:
Unfortunately the script I posted here on May 9 does not work any more, it fails at the first ``make``.
Just to be sure, I gave it a fresh try without the ``git checkout`` and ``sed`` commands, which compiles ok, but the hello-test fails.

So, currently, there seems no way any more to compile code for RV32I and simulate it together with pk in spike, other than to use the toolchain I compiled on May, 9.

emelcher, 2018-07-09:
Found a solution:
Omit the first ``git checkout`` in the script I posted on May 9.

LGKev, 2018-08-10:
currently trying this now. if it still works i'll report back. Thanks @emelcher !!