riscv-software-src/riscv-pk #41: Config string code looks like it was lost in a bad push?

sorear, 2016-11-04:
https://github.com/riscv/riscv-pk/commit/b2b484e324679a7c5b3a3744cf90b453ad7dae7a is now showing on zero branches, and https://github.com/riscv/riscv-pk/blob/master/machine/sbi.h no longer documents any way to access the config string.  It could be that the config string is deliberately no longer exposed to S-mode but I think it more plausible that the repository is broken.

aswaterman, 2016-11-04:
We do plan to expose a config string to Linux, but the contents and mechanism are still TBD (but must be resolved soon).  The commit you referenced was a temporary measure.

sorear, 2016-11-04:
@aswaterman ok.  since it was force-pushed away and not `git revert`ed, I thought it was a mistake.  If it's deliberately gone shall I close this?

aswaterman, 2016-11-04:
Yeah, not a mistake.. I'll close.

closed 2016-11-04 by aswaterman