riscv-software-src/riscv-pk #43: bbl always access "Invalid htif register address"

shrshore, 2016-12-08:
when I build this project to boot the 32-bit linux kernel with bbl, it always print out that “Invalid htif register address" and exit.
Here is my configuration of riscv-pk:
```
$ cd riscv-pk
$ mkdir build
$ cd build
$ ../configure --prefix=$RISCV --host=riscv32-unknown-linux-gnu --enable-32bit --with-payload=/home/shore/linux-4.6.2/vmlinux
$ make
$ make install
```
And when I run the bbl with payload of RV32 linux kernel on qemu 
```
$ cd ../riscv-qemu
$ ./riscv32-softmmu/qemu-system-riscv32 -nographic -kernel /home/shore/riscv/riscv32-unknown-elf/bin/bbl
```
I wonder how to fix this error to boot the linux kernel. Anyone can help?

sorear, 2016-12-14:
Worth asking at this point whether we should keep the old HTIF mechanism or move pk/bbl/spike over to using a virtual UART protocol; that might even be friendlier to physical harnessed test chips.

The HTIF is already (slowly) being phased out so I'm not terribly interested in supporting it more widely than it already is.  @sagark thoughts?

The error message you're getting is from QEMU; I don't know whether the fault is in pk or qemu

sorear, 2016-12-14:
Actually riscv-qemu doesn't seem to even build right now in a 32-bit chroot: @sagark 

```
/home/test/riscv-qemu/hw/riscv/sifive_board.c:215:5: error: format '%lx' expects argument of type 'long unsigned int', but argument 3 has type 'uint64_t' [-Werror=format=]
     sprintf(ramsize_as_hex_str, "%016lx", rsz);
```

@shrshore If you have a patch that fixes the build in a non-hacky way, that would be useful to discuss as a PR against qemu.

sorear, 2016-12-16:
OK, we have qemu building, but [the HTIF console is not implemented on 32-bit, it just traps if the device is set](https://github.com/riscv/riscv-pk/blob/master/machine/htif.h#L11).  Trying sifive_board now.

sorear, 2016-12-16:
1. Get riscv/riscv-gnu-toolchain@2ffda4e465 (this is a 1.9 version, as QEMU does not support 1.9.1)
2. `./configure --with-xlen=32 --prefix=$HOME/opt-32-1.9`, build. and add to PATH
3. sifive/riscv-pk@sifive (presently sifive/riscv-pk@fd688fc47c)
4. `../configure --host=riscv32-unknown-linux-gnu`
5. `make -k` ; pk fails with an gcc ICE on this toolchain version, but we only need bbl
6. Build qemu using https://github.com/riscv/riscv-qemu/pull/45 and the TARGET_LONG_BITS change from my comment
7. `riscv32-softmmu/qemu-system-riscv32 -machine sifive -nographic -kernel $HOME/riscv-pk-sifive/build/bbl`

This gets far enough as to execute the dummy_payload.  Haven't tried booting Linux.

aswaterman, 2016-12-16:
Nice.  I think the HTIF issue is EWONTFIX.  @a0u is working on the general-purpose UART, which we should standardize on in favor of the tethering stuff.

sagark, 2016-12-16:
Yeah, I'm all for removing the HTIF code from QEMU once the UART is stable and supported by the rest of the tools.

sorear, 2016-12-16:
(I'm still a little bit unclear on how the UART stuff overlaps with external debug for the tethered chips.  Would you need both?  The point of HTIF right now seems to be that you can run pk and sbicon over the same debug tether that you want anyway for run control.)

shrshore, 2016-12-20:
@sorear According to your procedures, I build a 32bit bbl on 64bit ubuntu. And the bbl can run correctly to print out the LOGO.
Then I build vmlinux according to https://github.com/Technolution/riscv-linux/tree/priv-1.9, and build the bbl with the vmlinux as payload, but when it comes to
```
riscv32-softmmu/qemu-system-riscv32 -machine sifive -nographic -kernel $HOME/riscv-pk-sifive/build/bbl
```
It meets Kernel Panic: not Syncing. Here is the output:
```
[    0.000000] Linux version 4.6.2-00032-g1ef29a9-dirty (dxu@ubuntu) (gcc version 6.1.0 (GCC) ) #1 Wed Dec 7 20:07:32 CST 2016
[    0.000000] Available physical memory: 124MB
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080400000-0x0000000087ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080400000-0x0000000087ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080400000-0x0000000087ffffff]
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 31496
[    0.000000] Kernel command line:
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 123216K/126976K available (1967K kernel code, 67K rwdata, 336K rodata, 60K init, 189K bss, 3760K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:0 nr_irqs:0 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.000000] pid_max: default: 32768 minimum: 301
[    0.000000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.290000] devtmpfs: initialized
[    0.300000] clocksource: jiffies: mask: 0xffffffff00000000 max_cycles: 0xffbc193000000000, max_idle_ns: -4601682592913955121 ns
[    0.320000] NET: Registered protocol family 16
[    0.370000] clocksource: Switched to clocksource riscv_clocksource
[    0.410000] NET: Registered protocol family 2
[    0.440000] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.470000] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.470000] TCP: Hash tables configured (established 1024 bind 1024)
[    0.470000] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.470000] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.480000] NET: Registered protocol family 1
[    1.080000] console [sbi_console0] enabled
[    1.110000] futex hash table entries: 256 (order: -1, 3072 bytes)
[    1.130000] workingset: timestamp_bits=29 max_order=15 bucket_order=0
[    2.200000] io scheduler noop registered
[    2.220000] io scheduler cfq registered (default)
[    3.080000] VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
[    3.110000] Please append a correct "root=" boot option; here are the available partitions:
[    3.140000] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    3.140000] CPU: 0 PID: 1 Comm: swapper Not tainted 4.6.2-00032-g1ef29a9-dirty #1
[    3.140000] Call Trace:
[    3.140000] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
```
Wondering how to fix it. Thanks a lot!

sorear, 2016-12-20:
There's no support for any block devices in qemu right now so you need to build the kernel with an initramfs.  https://github.com/sifive/freedom-u-sdk/blob/master/Makefile is what I'd follow (if you just clone that repository and run "make", you'll get a riscv64 kernel+bbl that can be used with qemu-system-riscv64; I have tested this and it works even on a 32-bit host)

MaMazav, 2017-05-08:
https://github.com/riscv/riscv-tools/issues/70 may help.