riscv-software-src/riscv-pk #44: problem compiling: 'GIGAPAGE_SIZE' undeclared (first use in this function)

manuelafm, 2017-01-12:
This happens with master.  Is it broken, or am I doing something wrong?
```
[...]
a - snprintf.o
a - string.o
riscv64-unknown-elf-ranlib libutil.a
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"dummy_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/pk.c
riscv64-unknown-elf-gcc -nostartfiles -nostdlib -static  -o pk pk.o -L.  -lpk  -lmachine  -lsoftfloat  -lutil -lgcc -T ../pk/pk.lds
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"dummy_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../bbl/bbl.c
../bbl/bbl.c: In function 'supervisor_vm_init':
../bbl/bbl.c:25:58: error: 'GIGAPAGE_SIZE' undeclared (first use in this function)
   size_t num_middle_pts = (-info.first_user_vaddr - 1) / GIGAPAGE_SIZE + 1;
                                                          ^~~~~~~~~~~~~
../bbl/bbl.c:25:58: note: each undeclared identifier is reported only once for each function it appears in
Makefile:300: recipe for target 'bbl.o' failed
make: *** [bbl.o] Error 1
```

aswaterman, 2017-01-12:
This would happen with an older gcc (prior to the introduction of the __riscv_xlen macro).

manuelafm, 2017-01-12:
Hi aswaterman,

You're right, I came back to suggest that maybe that was the problem.  Sorry for the noise!

closed 2017-01-12 by manuelafm