riscv-software-src/riscv-pk #47: segfault on binary with large static initialization data

nakengelhardt, 2017-03-15:
I have a small matrix multiply benchmark that previously used to run with spike pk. I installed the latest riscv-tools from github, but now it segfaults:

```
$ spike pk mm
z  0000000000000000 ra 000000000001022c sp 000000007ffc51a0 gp 0000000000039cc8
tp 0000000000000000 t0 0003c5230004131b t1 00000000800052b0 t2 0003a662000420a4
s0 000000007ffe2670 s1 0000000000009c40 a0 000000007ffec2b0 a1 000000007ffe2670
a2 000000007ffec2b0 a3 0000000000000000 a4 0000000000000024 a5 00000000800052b0
a6 000000007ffcef80 a7 000000008001e2b0 s2 fffffffffffcf000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000019000 t4 000000007ffe28f0 t5 00000000800052b0 t6 0003e6ec00033547
pc 000000000001046c va 00000000800052b0 insn       ffffffff sr 8000000000006000
User load segfault @ 0x00000000800052b0
```

The matrix data is defined in the code as a static initialization of three 160x160 matrices (const int a[160][160] = {...}; etc). I reduced the data size to 100x100 and then it runs correctly. My predecessors used to compile and run in 32bit mode whereas I installed the default 64bit tools, so that probably increased the space used by the static data.

Increasing the memory space available to spike using the -m flag has no effect.

Is this a bug or am I running into some size limit imposed on the input binary by pk?

poweihuang17, 2017-04-14:
I got the same problem with this code. Does anyone has some idea about what to deal with this now.
1 #define M 1000
  2 #define N 2000
  3 #include <stdio.h>
  4 int main(){
  5     int sum;
  6     int a[M][N];
  7     for (int i=0;i<M;i++)
  8         for (int j=0;j<N;j++)
  9             a[i][j]=i+j;
 10     printf("Matrix\n");
 11 }

kito-cheng, 2017-04-16:
@nakengelhardt Can you provide a testcase?

@poweihuang17 pk set stack size to 256 k, and you program require 1000 * 2000 * sizeof (int) = 16M, so your case is stack overflow.

[1] https://github.com/riscv/riscv-pk/blob/master/pk/mmap.c#L404

nakengelhardt, 2017-04-16:
Oh right. It's the stack for me too, the multiply function made a copy of the two input matrices to the stack for some reason.

closed 2017-04-16 by nakengelhardt

poweihuang17, 2017-04-16:
@kito-cheng  Thanks for your reply. But that setting (256K) will make pk pretty limited. I will try to use spike to run my program in bare-metal mode without pk now. However, I'm wondering why does people set such a small stack for pk? Isn't that a problem that we should modify?

kito-cheng, 2017-04-18:
@aswaterman How about increase stack size to 8M in `pk` just like x86/linux default?

aswaterman, 2017-04-18:
pk also runs on systems with very limited RAM, so 8 MiB is too large.  I guess we can give it something like 3% of memory, up to a max of 8 MiB.

aswaterman, 2017-04-18:
0f0ac64

theotuck, 2017-04-21:
even increasing the stack size, there is constantly user load segfault for me. 
for doing my own test, I even modify the stack size to 32768*RISCV_PGSIZE, but it doesn't work.
 
Here is my command line:
`root@osboxes:/home/osboxes/mycode/V7/router-trie/poptrie-master# spike pk poptrie_test_basic `

the output:
`init: .passed
lookup: start the poptrie_fib_ret 
z  0000000000000000 ra 0000000000014154 sp 000000007ffffa70 gp 000000000002c8c0
tp 0000000000000000 t0 0000000000017e10 t1 000000007ffff920 t2 0000000000000000
s0 000000007ffffaa0    s1 0000000000000000 a0 0000000000000013 a1 000000007ffff91c
a2 000000000002c283 a3 0000000000000400 a4 00000000028bc890 a5 00000000028c5000
a6 8080808080808080 a7 fefefefefefefeff         s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 fffffffffffffffb              t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 0000000000014168 va 00000000028c5000 insn       ffffffff              sr 8000000000006000
User load segfault @ 0x00000000028c5000
`
and the disassembler:
`   14160:	fec42783          	lw	a5,-20(s0)
   14164:	0792                	slli	a5,a5,0x4
   14166:	97ba                	add	a5,a5,a4
   14168:	639c                	ld	a5,0(a5)
   1416a:	fd043703          	ld	a4,-48(s0)
   1416e:	02f71263          	bne	a4,a5,14192 <poptrie_fib_ref+0x62>
`

theotuck, 2017-04-21:
even increasing the stack size, there is constantly user load segfault for me.
for doing my own test, I even modify the stack size to 32768*RISCV_PGSIZE, but it doesn't work.

Here is my command line:
root@osboxes:/home/osboxes/mycode/V7/router-trie/poptrie-master# spike pk poptrie_test_basic

the output:
`init: .passed lookup: start the poptrie_fib_ret 
z 0000000000000000 ra 0000000000014154 sp 000000007ffffa70 gp 000000000002c8c0 
tp 0000000000000000 t0 0000000000017e10 t1 000000007ffff920 t2 0000000000000000 
s0 000000007ffffaa0    s1 0000000000000000 a0 0000000000000013 a1 000000007ffff91c 
a2 000000000002c283 a3 0000000000000400 a4 00000000028bc890 **a5 00000000028c5000**
a6 8080808080808080 a7 fefefefefefefeff        s2 0000000000000000  s3 0000000000000000 
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000 
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000 
t3 fffffffffffffffb              t4 0000000000000000 t5 0000000000000000 t6 0000000000000000 
**pc 0000000000014168** **va 00000000028c5000** insn ffffffff                    sr 8000000000006000
User load segfault @ 0x00000000028c5000 `

and the disassembler:
`
14160: fec42783 lw a5,-20(s0) 
14164: 0792 slli a5,a5,0x4 
14166: 97ba add a5,a5,a4 
**14168: 639c ld a5,0(a5) **
1416a: fd043703 ld a4,-48(s0) 
1416e: 02f71263 bne a4,a5,14192 <poptrie_fib_ref+0x62>
`

theotuck, 2017-04-21:
even increasing the stack size, there is constantly user load segfault for me.
for doing my own test, I even modify the stack size to 32768*RISCV_PGSIZE, but it doesn't work.

Here is my command line:
root@osboxes:/home/osboxes/mycode/V7/router-trie/poptrie-master# spike pk poptrie_test_basic

the output:

init: .passed 
lookup: start the poptrie_fib_ret 
`
z 0000000000000000 ra 0000000000014154 sp 000000007ffffa70 gp 000000000002c8c0 tp 0000000000000000 t0 0000000000017e10 t1 000000007ffff920 t2 0000000000000000 s0 000000007ffffaa0 s1 0000000000000000 a0 0000000000000013 a1 000000007ffff91c a2 000000000002c283 a3 0000000000000400 a4 00000000028bc890 a5 00000000028c5000 a6 8080808080808080 a7 fefefefefefefeff s2 0000000000000000 s3 0000000000000000 s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000 s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000 t3 fffffffffffffffb t4 0000000000000000 t5 0000000000000000 t6 0000000000000000 pc 0000000000014168 va 00000000028c5000 insn ffffffff sr 8000000000006000 User load segfault @ 0x00000000028c5000 `

and the disassembler:

`14160: fec42783 lw a5,-20(s0)
14164: 0792 slli a5,a5,0x4 
14166: 97ba add a5,a5,a4 
14168: 639c ld a5,0(a5) 
1416a: fd043703 ld a4,-48(s0) 
1416e: 02f71263 bne a4,a5,14192 <poptrie_fib_ref+0x62> `

poweihuang17, 2017-04-21:
@theotuck  Could you upload your code so that we could debug?