riscv-software-src/riscv-pk #52: spike pk hello. could not open pk

ninode, 2017-04-06:
Hello,

 I know this has been asked so many times and I still can't resolve it following other responses.
I cloned my sources from the **rocket-chip** git.

Tthe toolchain built was successful (adding zlib).

 When I run **spike pk hello** it yields: 
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open pk
alpha@alpha-VirtualBox:~/rocke
```

So, I need guidance and instructions for setting up the environment variables (I'm new to linux). Also, I've noticed that building from the LowRISC repo worked for me as opposed to building from the _rocket-chip_ repo. in _LowRISC_ **spike pk hello** works fine after I source the _set_riscv_env file._ In _rocket-chip_ there isn't such file, and I don't know how I can set up the environment myself. 

Also, when I configured the _riscv-pk/build_ directory I ran **../configure --prefix=$RISCV/ --host=riscv64-unknown-elf**. Then  **../configure --prefix=$RISCV/riscv64-unknown-elf/bin/pk --host=riscv64-unknown-elf**. Both didn't work.

Does anyone know how I can fix this ? 

Thank you :)

closed 2017-09-17 by ninode

adamsir32, 2023-03-10:
hello，i have the same qustion，how to solve it？thank you

jundijiujieke, 2023-05-04:
Did you fix this problem, I already install the pk on my linux, but they can't find it. Please tell me how to fix this problem.