riscv-software-src/riscv-pk #54: Infinite trap_instruction_access_fault loop while calling sbi_hart_id during Linux startup

liujed, 2017-05-21:
While trying to get Linux 4.6.2 running in spike, I am getting a series of trap_instruction_access_faults that appear to be looping. In case I am doing something wrong, here is the sequence of commands that I used to clone and build riscv-tools and riscv-linux: https://pastebin.com/F0g5p8ZX. I am on the latest master for both repos (14bbe5a and 9079be6, respectively).

The first fault occurs when calling `sbi_hart_id` at `riscv-linux/arch/riscv/kernel/head.S:28`.
```
core   0: 0x0000000080200018 (0x800000e7) jalr    ra, zero, -2048
: 
core   0: exception trap_instruction_access_fault, epc 0xfffffffffffff800
core   0:           badaddr 0xfffffffffffff800
```
Stepping past the fault results in the pc getting sent to the start of `trap_vector` (`riscv-pk/machine/mentry.S:35`), which jumps to `.Lhandle_trap_in_machine_mode`, and results in a series of nested calls: `pmp_trap` → `redirect_trap` → `__redirect_trap` → `restore_mscratch`. At `mentry.S:185`, `restore_mscratch` does an `mret`, which results in a second fault:
```
core   0: 0x00000000800001a4 (0x30200073) mret
: 
core   0: exception trap_instruction_access_fault, epc 0x0000000000000000
core   0:           badaddr 0x0000000000000000
```
The pc gets sent back to `trap_vector` again, and goes through the same series of jumps and calls, ending up back at `mentry.S:185`. The `mret` faults again with the same address, which seems like an infinite loop:
```
core   0: 0x00000000800001a4 (0x30200073) mret
: 
core   0: exception trap_instruction_access_fault, epc 0x0000000000000000
core   0:           badaddr 0x0000000000000000
```

sorear, 2017-05-21:
riscv-tools master is privileged spec 1.10; so I _think_ you should be using the riscv-next branch of riscv-linux in order to have a matching SBI.

liujed, 2017-05-21:
Thanks for the quick reply. What branches should I be tracking if I want HTIF support? I just saw that a similar issue was reported in #18, and @zhemao's [comment](https://github.com/riscv/riscv-pk/issues/18#issuecomment-206115996) there suggests HTIF was dropped in priv-1.9. The comment says to track master for HTIF, but I guess this might have changed in the past year.

sorear, 2017-05-21:
The master branch has been 1.9 for several months now.

HTIF has referred to several rather different things over the past year so I'm not sure exactly what you're looking for.

liujed, 2017-05-21:
Sorry, I usually work several layers of abstraction above this. I guess I am referring to the mechanism by which the proxy kernel communicates with an FPGA host to forward syscalls. The thing that slide 12 is talking about in this slide deck: https://riscv.org/wp-content/uploads/2015/01/riscv-software-stack-bootcamp-jan2015.pdf. The goal being to set up communication between programs running on the host and the RISC-V target.