riscv-software-src/riscv-pk #56: Correct configure command for 32-bit pk

ibolton, 2017-05-31:
This is just a minor issue, but I've been trying to get spike working with 32-bit binaries and have found that supplying --enable-32bit here causes build failures (there's no -m32 option any more), but the following configure command does work (assuming you have a riscv32-unknown-elf toolchain built):
```
../configure --prefix=$RISCV --host=riscv32-unknown-elf
```
riscv-isa-sim can then be configured as follows:
```
../configure --prefix=$RISCV --with-fesvr=$RISCV --with-isa=rv32ima
```
If you could update the README.md for this repo (and maybe also riscv-isa-sim), I think that will help people.

neocogent, 2017-06-06:
I tried this too and had similar but confusing results. Even though building the 32-bit pk puts the results in a riscv64-uknown-elf directory. Turns out that works but I tried moving them to the 32 bit dir before figuring out it was actually right. Just not what I expected.

The correct configure lines are present in the build-rv32ima.sh script and I wanted to build a RV32i version (not ima) and so just adapted from that.

Also, you need to make fesvr before riscv-isa-sim. It all works in the end.

ibolton, 2017-06-06:
Hi neocogent. Sadly, I only found the build-rv32ima.sh script after battling with the above, but it was good to see that I arrived at the right answer independently. :) Hopefully the README.md here can still be updated to save people some confusion and trouble. Good luck with your RV32i project.

Quasilyte, 2017-06-17:
Facing the same issue.
I can't find `build-rv32ima.sh`, but removing `-m32` flags from `Makefile` solves an issue (at least build succeeds).

Quasilyte, 2017-06-18:
I can confirm that without `-m32` pk seems usable with spike.