riscv-software-src/riscv-pk #63: How to enable RoCC in BBL

phthinh, 2017-09-15:
I know the way to enable RoCC in pk, but when I boot linux with BBL. It seems that the RoCC is not enable.
Could anybody know how to enable RoCC in BBL?

seldridge, 2017-09-15:
This shouldn't be handled at the BBL level, but inside the kernel via context switch code. The quick and dirty hack is to set `SR_XS` whenever a process gets swapped in similar to how `SR_FS` works. Note: that doing this without saving and restoring the RoCC's state is going to result in bugs 

If you search for usages of `SR_FS*` (see: [`csr.h`](https://github.com/riscv/riscv-linux/blob/32b1573d1f118844d859341d095e005ca5ba572e/arch/riscv/include/asm/csr.h#L26)) you should get some idea of how to do this.

There's an [old thread about this over on riscv-linux with a much more detailed explanation](https://github.com/riscv/riscv-linux/issues/32).

phthinh, 2017-09-18:
Thanks for your hint! I have a small modification inside the kernel in the process.c file to enable RoCC. Custom instructions are able to be run with RoCC at the Linux userspace.

closed 2017-09-18 by phthinh