riscv-software-src/riscv-pk #64: Double-Checked Lock in

markww, 2017-10-20:
I was looking in mmap.c and noticed the following code:

```C
static vmr_t* __vmr_alloc(uintptr_t addr, size_t length, file_t* file,
                          size_t offset, unsigned refcnt, int prot)
{
  if (!vmrs) {
    spinlock_lock(&vm_lock);
      if (!vmrs)
        vmrs = (vmr_t*)__page_alloc();
    spinlock_unlock(&vm_lock);
  }
 /* ... */
```

The above is a straightforward example of the common double-checked-lock anti-pattern. In essence, the outer unlocked check of vmrs is used as a symbol to check to see whether vmrs has been initialized. Unfortunately, vmrs not

aswaterman, 2017-10-20:
The problem is just that there's no ordering between the call to __page_alloc and the assignment to vmrs.  I'll add a memory barrier.

aswaterman, 2017-10-20:
4c64de3213e7a08e4097970b26a14ad835cabfab

closed 2017-10-20 by aswaterman