riscv-software-src/riscv-pk #65: Incorrect result from single-precision FP emulation code

asb, 2017-10-31:
See https://github.com/riscv/riscv-isa-sim/issues/147 for earlier discussion.

```
$ cat t.c
#include <stdlib.h>

float s2f(int s) {
  return s;
}

int main() {
  if (s2f(0x80000000) != -2147483648.0)
    abort();
  return 0;
}

$ riscv32-unknown-elf-gcc -O0 -march=rv32imaf -mabi=ilp32f t.c
$ spike --isa=rv32ima pk a.out; echo $?
1
```

The emulated fcvt+fmv pair results in the incorrect value being generated in the a0 register in the `s2f` function body. Specifically, a0 contains `0xdf800000` rather than the expected `0xcf000000`. It's possible there's a spike emulation bug or a GCC bug, but the pk soft float code seems the most likely culprit at this time.

For reference, this is the assembly of the test program:

```
	.file	"t.c"
	.option nopic
	.text
	.align	2
	.globl	s2f
	.type	s2f, @function
s2f:
	add	sp,sp,-32
	sw	s0,28(sp)
	add	s0,sp,32
	sw	a0,-20(s0)
	lw	a5,-20(s0)
	fcvt.s.w	fa5,a5
	fmv.x.s	a0,fa5
	lw	s0,28(sp)
	add	sp,sp,32
	jr	ra
	.size	s2f, .-s2f
	.align	2
	.globl	main
	.type	main, @function
main:
	add	sp,sp,-16
	sw	ra,12(sp)
	sw	s0,8(sp)
	add	s0,sp,16
	li	a0,-2147483648
	call	s2f
	fmv.s.x	fa4,a0
	lui	a5,%hi(.LC0)
	flw	fa5,%lo(.LC0)(a5)
	feq.s	a5,fa4,fa5
	bnez	a5,.L4
	call	abort
.L4:
	li	a5,0
	mv	a0,a5
	lw	ra,12(sp)
	lw	s0,8(sp)
	add	sp,sp,16
	jr	ra
	.size	main, .-main
	.section	.rodata
	.align	2
.LC0:
	.word	3472883712
	.ident	"GCC: (GNU) 8.0.0 20171025 (experimental)"
```

aswaterman, 2017-10-31:
Pretty sure this is a result of two's complement overflow negating -2^31.  Today I don't have access to a machine that would allow me to test this patch, so can I volunteer you?  https://github.com/riscv/riscv-pk/pull/66

aswaterman, 2017-11-01:
Resolved by #66

closed 2017-11-01 by aswaterman