riscv-software-src/riscv-pk #78: build fails on arch linux

rain-1, 2018-02-01:
Hello, I am on x86_64 GNU/Linux arch linux.

I did`../configure --host=riscv64-unknown-elf` but

```
$ make
gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/file.c
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
make: *** [Makefile:302: file.o] Error 1
```

sorear, 2018-02-01:
You seem to be trying to build pk/bbl with an x86_64 compiler.  Was riscv64-unknown-elf-gcc in PATH at the time of the configure?

rain-1, 2018-02-01:
ah I understand, I didn't realize I needed a riscv toolchain to build this. Thank you.

closed 2018-02-01 by rain-1

Crosshairs, 2019-08-24:
> You seem to be trying to build pk/bbl with an x86_64 compiler. Was riscv64-unknown-elf-gcc in PATH at the time of the configure?

But  riscv64-unknown-elf-gcc in PATH at the time of the configure.Why still have the error:unrecognized argument in option ‘-mcmodel=medany’