riscv-software-src/riscv-pk #80: Support clone system call in riscv-pk

qtt2, 2018-02-08:
Hi,

I plan to support clone() system call to create and map threads to cores in riscv-pk. Basically I want pk to make a copy of current thread context that calls clone(), make a new thread context and start the new thread on a new core. Can anyone give me some directions for implementing this system call in pk? Thank you very much!

Tuan

aswaterman, 2018-02-08:
It will require major changes, as pk isn't even aware of the concept of multiple user threads.  (There's only one context structure, there's no scheduler, etc.)

I'd suggest you just use Linux if you want to run pthreads applications.

anycore, 2018-03-29:
Is there a HOWTO page or some documentation on how to run pthread applications on Linux using Spike? Is it even possible? Any pointers is highly appreciated.