riscv-software-src/riscv-pk #83: Calculation of Enable Offsets for PLIC

LeeMooreImperas, 2018-02-16:
Hi

I am struggling to understand the logic of the address calculation for the PLIC setup
in machine/fdt.c we have the following defines
```
#define HART_BASE	0x200000
#define HART_SIZE	0x1000
#define ENABLE_BASE	0x2000
#define ENABLE_SIZE	0x80
```
These defines are used later on to define the Machine and Supervisor offsets for the HARTS.
if HART-0 has Machine and Supervisor mode, I would expect these registers to appear at (referenced to HART_BASE)
```
0x202000 - 0x20201F (Hart-0 Mode-M Interrupt Enables 0-1023)
0x202020 - 0x20203F (Hart-0 Mode-S Interrupt Enables 0-1023)

```
However, in the address calculations performed this actually appears as 

```
0x202000 - 0x20201F (Hart-0 Mode-M Interrupt Enables 0-1023)
0x202080 - 0x20209F (Hart-0 Mode-S Interrupt Enables 0-1023)

```
this is because the address calculation uses index to multiply the ENABLE size of 0x80

I would have thought the enable size of 0x80 is for the entire set of HART enables for all modes, ie Machine, Supervisor (and futureproof Hypervisor, User)
The existing address calculation seems to make the layou look as follows

```
0x202000 - 0x20201F (Hart-0 Mode-M Interrupt Enables 0-1023)
0x202080 - 0x20209F (Hart-1 Mode-M Interrupt Enables 0-1023)
```

Have I got this completely wrong ?

Thx
Lee

LeeMooreImperas, 2018-02-16:
@palmer-dabbelt 
Hi Palmer,
I saw you wrote the code related to my question, and it seems the same code on the Linux side drivers/irqchip/irq-riscv-plic.c
do you think this could be wrong on both sides, ie the boot loader and the Linux ?

Thx
Lee

sorear, 2018-02-17:
I don't actually follow the question, but have you looked at [the documented memory map for the sifive PLIC](https://static.dev.sifive.com/U54-MC-RVCoreIP.pdf#page=40) ?

LeeMooreImperas, 2018-02-19:
@sorear  
Thanks for the reply.
I have looked at the Memory maps for all of the PLIC implementations, which is why I cannot understand what I am seeing.

As far as I can see the PLIC has a region of enables for each HART starting at an offset of 0x2000
each HART has 0x80 enables which are split into 4 separate banks of 0x20 for Machine, Supervisor, and currently un-implemented User and Hypervisor

So if HART-0 was setup to have Machine and Supervisor support, I would expect to see registers appearing at 
0x2000 - 0x201F : HART-0 Machine Mode Enables
0x2020 - 0x203F : HART-0 Supervisor Mode Enables
0x2040 - 0x207F : HART-0 (Unimplemented space for Hypervisor/User)
0x2080 - 0x209F : HART-1 Machine Mode Enables ....

Does this match your understanding ?

If so, then something not quite correct is happening.
When I boot the Linux (under Qemu) for a virtio board, the configuration is for HART-0 to have Machine and Supervisor Enables.
When I see an access to the HART-0 Supervisor enables, the access address is 0x2080, which is the HART-1 Machine Mode enables.

So it appears the documentation does not match the execution.
I am pretty confused by what I am seeing

Thx
Lee

sorear, 2018-02-19:
No, each hart has 0x80 bytes of enables (1023 interrupts packed 8 per byte) *per mode*.  And there's no banking or "unimplemented space".  0x80 gets you from one hart/mode combination to the next hart/mode combination, and hart/mode combinations are densely packed.  The U54-MC has 1 M-only hart and 4 M+S harts, so it has 9*0x80 bytes of enables.

LeeMooreImperas, 2018-02-19:
@sorear 
I see the issue I was misunderstanding now.
So to confirm, if I had 4 HARTS in the following configuration
M, M, M, M
the (base) offset addresses would be as follows
0x2000 Hart-0 Machine Mode
0x2080 Hart-1 Machine Mode
0x2100 Hart-2 Machine Mode
0x2180 Hart-3 Machine Mode

If I had 4 HARTS with the following configuration
MS, MS, MS, MS
the (base) offset addresses would be as follows
0x2000 Hart-0 Machine Mode
0x2080 Hart-0 Supervisor Mode
0x2100 Hart-1 Machine Mode
0x2180 Hart-1 Supervisor Mode
0x2200 Hart-2 Machine Mode
0x2280 Hart-2 Supervisor Mode
0x2300 Hart-3 Machine Mode
0x2380 Hart-3 Supervisor Mode

It seemed strange to me that the location of HART-1 Machine Registers would be impacted by the modes supported by HART-0 (andevery subsequent HART thereafter)

Please confirm what I am saying above is correct.

Thx
Lee

sorear, 2018-02-19:
Your interpretation is correct.

palmer-dabbelt, 2018-02-21:
I'm going to close this as it seems everyone is on the same page, if I'm wrong then feel free to re-open it.

closed 2018-02-21 by palmer-dabbelt