riscv-software-src/riscv-pk #131: Problem during 32bit compilation

aignacio, 2018-11-06:
Hi,
I'm trying to build the pk/bbl to a 32-bit target but I think that the right argument it's --with-arch=rv32ima instead of --enable-32bit, also the libs like gcc are not pointing to the right path in the riscv-gnu-toolchain after installed.

Jesse-Millwood, 2019-07-23:
What I did was first put my cross compiler at the front of my `$PATH` and then defined the cross compilers as variables because I am using the multilib one: 

```
mkdir build/
cd build/

../configure --host=riscv32-unknown-elf CC=riscv64-unknown-elf-gcc CXX=riscv64-unknown-elf-g++ AR=riscv64-unknown-elf-ar READELF=riscv64-unknown-elf-readelf OBJCOPY=riscv64-unknown-elf-objcopy LDFLAGS="-march=rv32i -mabi=ilp32" CFLAGS="-march=rv32i -mabi=ilp32" --with-payload=${PAYLOAD} --with-arch=rv32i

make
``` 

If I run on qemu I can get to the first printm call but it hangs there.

aignacio, 2019-07-24:
tks @Jesse-Millwood

Jesse-Millwood, 2019-07-24:
It seems that the device tree needs to be at an aligned address after the payload? I'm not sure how to do this though.