riscv-software-src/riscv-pk #142: Disabling floating point with writing misa?

Hoblovski, 2019-01-23:
[These code](https://github.com/riscv/riscv-pk/blob/master/machine/minit.c#L76) tries to disable FPU by clearing the 'F' and 'D' bits in misa.

But according to the spec of v1.10 privileged, these bits in misa are WARL.
Thus clearing those bits might not take effect.
When it does not, the assert below fails.

Do you have any special reasons for that `assert(!(read_csr(misa) & fd_mask));`?

ps. The assert was introduced in [a commit three years ago](https://github.com/riscv/riscv-pk/commit/8006d00#diff-c1e089d3a9475c36046187ccdb062dbdR66)

aswaterman, 2019-01-23:
Take a look at the code you mentioned; the ifdef above is what I mean. Depending on which value for -march you pass GCC, it will (or will not) define __riscv_flen.

Hoblovski, 2019-01-23:
oops I see that. Thank you

closed 2019-01-23 by Hoblovski