riscv-software-src/riscv-pk #147: issues with hardened toolchain

nee-san, 2019-03-01:
Hi! I have issues with risc v bootloader. I've built my own toolchain for rv32imafdc with default PIE, PIC, SSP, RELRO support. I'm using musl libc for that. So there is log:
```
riscv32-linux-musl-ranlib libmachine.a
riscv32-linux-musl-gcc -Wl,--build-id=none -nostartfiles -nostdlib -static  -march=rv32imafdc -mabi=ilp32 -o pk pk.o -L.  -lpk  -lmachine  -lsoftfloat  -lutil -lgcc -T ../pk/pk.lds
riscv32-linux-musl-gcc -Wl,--build-id=none -nostartfiles -nostdlib -static  -march=rv32imafdc -mabi=ilp32 -o bbl bbl.o -L.  -lbbl  -ldummy_payload  -lmachine  -lsoftfloat  -lutil -lgcc -T ../bbl/bbl.lds
/home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/../../../../riscv32-linux-musl/bin/ld: /home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/libgcc.a(_divdi3.o): can't link double-float modules with soft-float modules
/home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/../../../../riscv32-linux-musl/bin/ld: failed to merge target specific data of file /home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/libgcc.a(_divdi3.o)
/home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/../../../../riscv32-linux-musl/bin/ld: /home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/libgcc.a(_moddi3.o): can't link double-float modules with soft-float modules
/home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/../../../../riscv32-linux-musl/bin/ld: failed to merge target specific data of file /home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/libgcc.a(_moddi3.o)
/home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/../../../../riscv32-linux-musl/bin/ld: /home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/libgcc.a(_udivdi3.o): can't link double-float modules with soft-float modules
/home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/../../../../riscv32-linux-musl/bin/ld: failed to merge target specific data of file /home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/libgcc.a(_udivdi3.o)
/home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/../../../../riscv32-linux-musl/bin/ld: /home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/libgcc.a(_umoddi3.o): can't link double-float modules with soft-float modules
/home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/../../../../riscv32-linux-musl/bin/ld: failed to merge target specific data of file /home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/libgcc.a(_umoddi3.o)
/home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/../../../../riscv32-linux-musl/bin/ld: /home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/libgcc.a(_clz.o): can't link double-float modules with soft-float modules
/home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/../../../../riscv32-linux-musl/bin/ld: failed to merge target specific data of file /home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/libgcc.a(_clz.o)
/home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/../../../../riscv32-linux-musl/bin/ld: read-only segment has dynamic relocations
collect2: error: ld returned 1 exit status
Makefile:319: recipe for target 'pk' failed
make: *** [pk] Error 1
make: *** Waiting for unfinished jobs....
/home/yuyu/zorro/OUT.riscv32/tools/lib/gcc/riscv32-linux-musl/8.3.0/../../../../riscv32-linux-musl/bin/ld: read-only segment has dynamic relocations
collect2: error: ld returned 1 exit status
Makefile:319: recipe for target 'bbl' failed
make: *** [bbl] Error 1
```
Configure options for riscv-pk:
```
unset CFLAGS CXXFLAGS LDFLAGS
export CFLAGS="-fno-stack-protector -fPIC"
export CXXFLAGS="$CFLAGS"
cd "$SRC"/riscv-pk-$version
mkdir build
cd build
../configure --host=riscv32-linux-musl --with-arch=rv32imafdc --with-abi=ilp32d --prefix='' --with-payload="/home/yuyu/zorro/OUT.riscv32/rootfs/boot/vmlinuz-linux"
make
```
Configure options for GCC:
```
Configured with: ../configure --prefix=/home/yuyu/zorro/OUT.riscv32/tools --libdir=/home/yuyu/zorro/OUT.riscv32/tools/lib --libexecdir=/home/yuyu/zorro/OUT.riscv32/tools/lib --build=x86_64-cross-linux-gnu --host=x86_64-cross-linux-gnu --target=riscv32-linux-musl --with-arch=rv32imafdc --with-linker-hash-style=gnu --with-pkgversion='januslinux edition 8.3.0 for riscv32' --with-bugurl=https://github.com/JanusLinux/janus/issues --with-sysroot=/home/yuyu/zorro/OUT.riscv32/rootfs --with-local-prefix=/home/yuyu/zorro/OUT.riscv32/rootfs --with-native-system-header-dir=/usr/include --with-isl --with-system-zlib --enable-__cxa_atexit --enable-checking=release --enable-clocale=generic --enable-default-pie --enable-default-ssp --enable-languages=c,c++,objc,obj-c++,fortran --enable-libstdcxx-time --enable-linker-build-id --enable-lto --enable-shared --enable-threads=posix --enable-tls --disable-gnu-indirect-function --disable-libmpx --disable-libmudflap --disable-libsanitizer --disable-libssp --disable-libstdcxx-pch --disable-multilib --disable-nls --disable-symvers --disable-werror
```

aswaterman, 2019-03-01:
I think there are two issues here:

- First, the "can't link double-float modules with soft-float modules" error typically results from trying to use an `-march` or `-mabi` setting that doesn't match the one you built libgcc for.  Make sure your compiler is built to generate `-march=rv32imafdc -mabi=ilp32` code.

- Second, pk is not general-purpose application code.  It's simply not going to work as a PIE.  PIEs need a loader for runtime relocation, but pk runs as bare-metal code; there's no loader to do that stuff.

nee-san, 2019-03-01:
Oh, okay. Thanks for info!

nee-san, 2019-03-02:
Now, I have this error log:
```
/home/yuyu/janus-private/OUT.riscv64/tools/lib/gcc/riscv64-linux-musl/8.3.0/../../../../riscv64-linux-musl/bin/ld: read-only segment has dynamic relocations
collect2: error: ld returned 1 exit status
Makefile:319: recipe for target 'pk' failed
make: *** [pk] Error 1
make: *** Waiting for unfinished jobs....
/home/yuyu/janus-private/OUT.riscv64/tools/lib/gcc/riscv64-linux-musl/8.3.0/../../../../riscv64-linux-musl/bin/ld: read-only segment has dynamic relocations
collect2: error: ld returned 1 exit status
Makefile:319: recipe for target 'bbl' failed
make: *** [bbl] Error 1
```

nee-san, 2019-03-02:
I used those *FLAGS:
```
export CFLAGS="${CFLAGS/-D_FORTIFY_SOURCE=2/}"
export CFLAGS="${CFLAGS/ -fstack-protector-strong/}"
export CFLAGS="${CFLAGS/ -Wl,-z,relro,-z,now/}"
export CFLAGS="-fno-PIE -fno-stack-protector -Wl,-z,norelro,-z,nonow -emit-relocs -fPIC $CFLAGS"
export CXXFLAGS="$CFLAGS"
```

aswaterman, 2019-03-02:
Sounds related to the hardened toolchain again. Try -fno-pic

nee-san, 2019-03-02:
Same problem

aswaterman, 2019-03-02:
Try a non-hardened toolchain?

nee-san, 2019-03-02:
My distro support hardened toolchain only, sorry.

aswaterman, 2019-03-02:
The problem is that there should be no dynamic relocations at all when compiling this program. Since I don’t have a similar toolchain, I can’t help you determine which flags you need to use to prevent their emission.

nee-san, 2019-03-03:
If you need hardened toolchain you can pass `--with-pic --enable-relro` for binutils and `--enable-default-pie --enable-default-ssp` for gcc

nee-san, 2019-03-06:
anyway try to fix it please

aswaterman, 2019-03-06:
Sorry, I don’t time to do that. As I said, it’s embedded software that is meant to be compiled with an embedded ELF toolchain, not a hardened Linux toolchain. If you or someone else figures it out, please let us know.

@jim-wilson any quick suggestions?

jim-wilson, 2019-03-06:
Linux and embedded elf toolchains aren't interchangeable.  Use an embedded elf toolchain to compile riscv-pk.

nee-san, 2019-03-06:
@jim-wilson thanks for advice!