riscv-software-src/riscv-pk #148: fail to trap in external interrupt

LoveZJT, 2019-03-21:
Hi, I want to study exception and interrupt of RISCV. And in pk, I change some codes to prompt me that it is trapping in interrupt. I change `handle_interrupt` function in [handlers.c](https://github.com/riscv/riscv-pk/blob/master/pk/handlers.c) as follow.
```
static void handle_interrupt(trapframe_t* tf)
{
  printk(".\n");  // I add this code to prompt me
  clear_csr(sip, SIP_SSIP);
}
```
And I write helloworld.c, which fall into an infinite loop finally.
```
#include <stdio.h>
int main()
{
  printf("Hello, World!\n");
  while(1);
}
```
After recompiling, I run `spike pk helloworld`. But I found that no `.` is printed. Even if I knock on the keyboard, trying to generate external interrupts, I still fail to see `.`.
I read the code and know that the handler of some exception and interrupts have been delegated to S-mode. After storing registers, it will jump to `handle_trap` function. If the cause is interruption, it will jump to handle_interrupt function to clear SIP_SSIP. Reading [riscv-priviledge-v1.10](https://github.com/riscv/riscv-isa-manual/blob/master/release/riscv-privileged-v1.10.pdf), I found that I/O interrupt belongs to external interrupts. 
Now I feel confused. When I knock on the keyboard, it will generate external interrupt, which should be handled in `handle_trap` function. But I can't see `.` owing to external interrupt. Have I got this completely wrong ?
Thx

aswaterman, 2019-03-21:
Keystrokes in Spike do not send interrupts to the target machine.  In fact, since Spike is not a full-system simulator, it doesn't support many interrupts at all.

Spike supports Machine Timer Interrupts and Machine Software Interrupts.  The former is (obviously) generated by the timer, and the latter is used for interprocessor interrupts, so is generated by software running on the target machine.

LoveZJT, 2019-03-21:
Thx. 
Riscv-qemu can simulate SiFive U Series, but I can't sure that it is a full-system simulator. Could you recommend a simulator that can support external interrupt? 
And it seems that only after initialing PLIC can it trap in external interrupt. Could you recommend some code or manuals about initialing PLIC ?
Thx again.

closed 2019-04-06 by LoveZJT