riscv-software-src/riscv-pk #149: Adding support for new physical to virtual memory translation

Bhagyaraja, 2019-03-21:
Hello,

I'm using this Proxy Kernel for Ariane. It is working fine with the existing peripherals of the Ariane. However, I'm unable to access the new peripherals (which have a new physical address) of the Ariane. Is there any way to add physical memory - virtual memory translation for the new peripherals (Similar to mmap() in Linux)?

aswaterman, 2019-03-21:
pk isn’t designed to support direct I/O access to userspace. In fact, it really is only meant to run very basic programs for testing purposes. You might be better off running Linux and using mmap, or writing device drivers.

Bhagyaraja, 2019-03-21:
Hello @aswaterman, 

Thank you for the quick response. Do you have any sample driver (that is used with proxy-kernel) for reference?

Regards,
Bhagyaraja Adapa

aswaterman, 2019-03-21:
Oh, sorry, a poorly placed comma made what I wrote ambiguous.  I meant, "running Linux and {either using mmap or writing device drivers}" - not writing a proxy kernel device driver.  It isn't architected to support real devices.  That is the "proxy" part of "proxy kernel" - it only supports fake devices by forwarding I/O to a host system.

Bhagyaraja, 2019-03-21:
Okay. Thank you for your response.

closed 2019-03-21 by Bhagyaraja