riscv-software-src/riscv-pk #153: Packaging riscv-pk

xyproto, 2019-04-04:
Hi,

Is there a command I can run to check if `riscv-pk` and/or `spike` has been installed correctly to the system?

With `spike /usr/riscv64-linux-gnu/bin/pk hello`, where `hello` is built from this source:

```c
#include <stdio.h>
#include <stdlib.h>

int main(int argc, char *argv[])
{
    printf("Hello, World!\n");
    return EXIT_SUCCESS;
}
```

With this command:

    riscv64-linux-gnu-gcc -O2 -o hello main.c

I get this error when running it:

```
spike /usr/riscv64-linux-gnu/bin/pk hello
bbl loader
z  0000000000000000 ra 00000000000103bc sp 000000007f7e9b50 gp 0000000000012800
tp 0000000000000000 t0 0000000000000000 t1 0000000000000000 t2 0000000000012360
s0 0000000000000000 s1 0000000000000000 a0 00000000000103a0 a1 0000000000000001
a2 000000007f7e9b58 a3 0000000000010462 a4 00000000000104ba a5 0000000000000000
a6 000000007f7e9b50 a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 ffffffffffffffff t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc fffffffffffffffe va fffffffffffffffe insn       ffffffff sr 8000000200046020
User fetch segfault @ 0xfffffffffffffffe
```

This is on 64-bit Arch Linux.

aswaterman, 2019-04-04:
Try building hello with the -static flag.

xyproto, 2019-04-05:
It works when building hello with `-static`! Thanks.

closed 2019-04-05 by xyproto