riscv-software-src/riscv-pk #154: Errors when compiling pk using 32bits compiler

hyf6661669, 2019-04-05:
The 32bits compiler (riscv-gnu-toolchain) is configured with ```--with-arch=rv32gc --with-abi=ilp32d```.

PK is configured with ```--host=riscv32-unknown-elf```.

And I have alreay added rv32gc_compiler to ```PATH```. However, the error occurs:
```
riscv32-unknown-elf-ranlib libmachine.a
riscv32-unknown-elf-gcc -Wl,--build-id=none -nostartfiles -nostdlib -static  -march=rv32gc -mabi=ilp32 -o pk pk.o -L.  -lpk  -lmachine  -lsoftfloat  -lutil -lgcc -T ../pk/pk.lds
riscv32-unknown-elf-gcc -Wl,--build-id=none -nostartfiles -nostdlib -static  -march=rv32gc -mabi=ilp32 -o bbl bbl.o -L.  -lbbl  -ldummy_payload  -lmachine  -lsoftfloat  -lutil -lgcc -T ../bbl/bbl.lds
/home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/../../../../riscv32-unknown-elf/bin/ld: /home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/libgcc.a(_divdi3.o): can't link hard-float modules with soft-float modules
/home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/libgcc.a(_divdi3.o)
/home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/../../../../riscv32-unknown-elf/bin/ld: /home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/libgcc.a(_moddi3.o): can't link hard-float modules with soft-float modules
/home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/libgcc.a(_moddi3.o)
/home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/../../../../riscv32-unknown-elf/bin/ld: /home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/libgcc.a(_udivdi3.o): can't link hard-float modules with soft-float modules
/home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/libgcc.a(_udivdi3.o)
/home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/../../../../riscv32-unknown-elf/bin/ld: /home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/libgcc.a(_umoddi3.o): can't link hard-float modules with soft-float modules
/home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/libgcc.a(_umoddi3.o)
/home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/../../../../riscv32-unknown-elf/bin/ld: /home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/libgcc.a(_clz.o): can't link hard-float modules with soft-float modules
/home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /home/hyf/rv32_compiler/rv32gc/lib/gcc/riscv32-unknown-elf/8.2.0/libgcc.a(_clz.o)
collect2: error: ld returned 1 exit status
Makefile:319: recipe for target 'pk' failed
make: *** [pk] Error 1
```

hyf6661669, 2019-04-05:
Besides, it seems that ```--enable-32bit``` option is removed from the latest version, I suggest modifying README.MD

s-macke, 2019-05-07:
I get very similar error messages such as 
```
riscv32-unknown-linux-gnu-gcc -Wl,--build-id=none -nostartfiles -nostdlib -static  -march=rv32g -mabi=ilp32 -o pk pk.o -L.  -lpk  -lmachine  -lsoftfloat  -lutil -lgcc -T ../pk/pk.lds
..... toolchain32/lib/gcc/riscv32-unknown-linux-gnu/8.3.0/../../../../riscv32-unknown-linux-gnu/bin/ld: .... toolchain32/lib/gcc/riscv32-unknown-linux-gnu/8.3.0/libgcc.a(_divdi3.o): can't link double-float modules with soft-float modules

```

aswaterman, 2019-05-07:
I think you need ilp32 libgcc.  By default, an rv32gc compiler will use the ilp32d ABI.

Alternatively, it might be possible to build pk with the ilp32d ABI; testing that would require some Makefile hacking.

hyf6661669, 2019-05-10:
Thanks, I will try to build the 32-bit compiler using ```--with-abi=ilp32```

s-macke, 2019-05-10:
Also works for me after I removed the d in ilp32d.

aswaterman, 2019-05-10:
@palmer-Dabbelt how best to manage this? I guess we could update the readme to say that an ilp32 or lp64 compiler is required and leave it at that.

hyf6661669, 2019-05-11:
In https://github.com/riscv/riscv-pk/blob/master/Makefile.in#L57-L61:
```
ifneq (@WITH_ARCH@,)
  march := -march=@WITH_ARCH@
  is_32bit := $(findstring 32,$(march))
  mabi := -mabi=$(if $(is_32bit),ilp32,lp64)
endif
```
At present, it seems only ```ilp32``` and ```lp64``` could be used to compile ```riscv-pk```.

hyf6661669, 2019-05-17:
@palmer-dabbelt 
Can we use ```ilp32/ilp32f/ilp32d/lp64/lp64f/lp64d``` to compile riscv-pk?

palmer-dabbelt, 2019-05-17:
Does this actually break with the hard-float ABIs?  I thought there was a reason for this but I can't actually find it.  I can still boot with

````
$ git diff | cat
diff --git a/Makefile.in b/Makefile.in
index 376670d2f449..d96f5d3fc3f3 100644
--- a/Makefile.in
+++ b/Makefile.in
@@ -57,7 +57,7 @@ endif
 ifneq (@WITH_ARCH@,)
   march := -march=@WITH_ARCH@
   is_32bit := $(findstring 32,$(march))
-  mabi := -mabi=$(if $(is_32bit),ilp32,lp64)
+  mabi := -mabi=$(if $(is_32bit),ilp32,lp64d)
 endif
 
 # Installation directories
````

If that's the problem then we can just fix it with autoconf.

aswaterman, 2019-05-17:
Since there's no floating-point argument passing in this code, it shouldn't make any difference.

(I wonder if we're just getting lucky about the compiler not spilling integers into FP registers when there's register pressure.  But that's unrelated to the ABI choice, since both ABIs have caller-saved FP registers when FP is enabled in -march.)

omerguzelelectronicguy, 2022-10-18:
In my case, it worked with `../configure --prefix=/home/omerguzel/Projects/compiler/rv32gcv --host=riscv32-unknown-elf --with-arch=rv32gcv --with-abi=ilp32d` for me. I used `rvv-next` branch.

omerkarslioglu, 2023-10-02:
After the deleting build directory and again creating build folder, try again:

../configure --prefix=$RISCV --host=riscv32-unknown-elf --with-arch=rv32gcv --with-abi=ilp32d

If it doesn't work try again with (without "d" extension):

../configure --prefix=$RISCV --host=riscv32-unknown-elf --with-arch=rv32gcv --with-abi=ilp32