riscv-software-src/riscv-pk #155: How does one build a standalone bbl?

marcopeereboom, 2019-04-11:
I read through the doco and searched the web but I could not find the magic to build a bbl that can be used to boot qemu with vmlinux instead of linking the bbl+kernel together.

Any clues?

davidlt, 2019-04-12:
IIRC upstream QEMU doesn't support separate BBL (firmware) and vmlinuz (kernel) files, but patches were posted in 2018 May, I think.

I would advice you to use OpenSBI instead ( https://github.com/riscv/opensbi ) which also has jump variant. Then modify QEMU line accordingly:
```
	-kernel build/platform/qemu/virt/firmware/fw_jump.elf \
	-device loader,file=<linux_build_directory>/arch/riscv/boot/Image,addr=0x80400000
```

There are more documentation and examples on OpenSBI GitHub page.

closed 2019-04-13 by marcopeereboom

marcopeereboom, 2019-04-14:
Yep this works like a charm. It would be helpful to send people to OpenSBI and deprecate this repo.