riscv-software-src/riscv-pk #174: While buidling BBL on mac hit - error: invalid value 'medany' in '-mcode-model medany'

htable579, 2019-09-10:
Hi, 

I am following https://buildmedia.readthedocs.org/media/pdf/risc-v-getting-started-guide/latest/risc-v-getting-started-guide.pdf 

I am at the step : 
Build BBL:
cd riscv-pk
mkdir build && cd build
../configure --enable-logo --host=riscv64-unknown-elf --with-payload=../../linux/vmlinux
make <<<< Hit the error when trying to build 

error: invalid value 'medany' in '-mcode-model medany' . <<<< Error 

 System Software Overview:
      System Version: macOS 10.14.6 (18G87)
      Kernel Version: Darwin 18.7.0

Please let me know, how to resolve this issue. Thanks !
regards,
Horst.

clion0003, 2019-10-31:
hi,
check whether risc-v gcc toolchain is in your $PATH and could you run `riscv64-unknown-elf-g++` properly. The  "-mcode-model" option is one of the options in risc-v gcc but not in clang. 

> Hi,
>
> I am following https://buildmedia.readthedocs.org/media/pdf/risc-v-getting-started-guide/latest/risc-v-getting-started-guide.pdf
>
> I am at the step :
> Build BBL:
> cd riscv-pk
> mkdir build && cd [...]

kuzminrobin, 2019-12-13:
This needs to be explicitly mentioned in README.md. I had a similar issue.

OddLingo, 2021-06-03:
I ran into this.  I did have the GNU toolchain in my PATH but it was named 'riscv64-linux-gnu-*' and not 'riscv64-unknown-linux-gnu-*'.  Maybe they changed the name since the README.md was written?  Changing the --host value on configure fixed it.