riscv-software-src/riscv-pk #175: Cannot simulate Linux with spike successfully

Liboya-CB, 2019-09-19:
Hi,

I am trying to simulate latest Linux with spike on Ubuntu 16.04. But it seems that the device tree cannot be initialized successfully.

The version of related repositories as listed as belows:
1. **riscv-gnu-toolchain**: [7833a53](https://github.com/riscv/riscv-gnu-toolchain/tree/7833a53f8b0d0edb2bec0bb9a177685ae75570d5)
2. **riscv-pk**: [a3e4ac6](https://github.com/riscv/riscv-pk/tree/a3e4ac61d2b1ff37a22b9193b85d3b94273e80cb)
3. **riscv-isa-sim**: [88a8528](https://github.com/riscv/riscv-isa-sim/tree/88a852836acb4c7166b1aa4102e11354bfd99234)
4. **linux**: [v5.3-rc4](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=d45331b00ddb179e291766617259261c112db872)
5. **busybox**: [ http://busybox.net/downloads/busybox-1.26.2.tar.bz2]( http://busybox.net/downloads/busybox-1.26.2.tar.bz2)

After compiling the toolchain, I build the vmlinux and bbl with these commands respectively:

```shell
# vmlinux
make -j16 ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu-
```

```shell
# bbl
 ../configure --prefix=$RISCV --host=riscv64-unknown-linux-gnu --with-payload=/home/my/Riscv/linux/vmlinux
```

After executing ***spike bbl***,  no information is shown except an error.
> ERROR: invalid load from debug module: 1 bytes at 0x0000000000000101

This error information is reported in `fdt_check_header()` function. That is to say, the console is not initialized as well. 

I listed the address of device tree pointer in `enter_supervisor_mode` from[`machine/minit.c`](https://github.com/riscv/riscv-pk/blob/a3e4ac61d2b1ff37a22b9193b85d3b94273e80cb/machine/minit.c), which is the newset value before starting kernel. 
With the support of SBI function (`sbi_console_putchar()`), I listed the address of device tree pointer in `setup_vm` in [`arch/riscv/mm/init.c`](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/riscv/mm/init.c?id=d45331b00ddb179e291766617259261c112db872), which is the first use of these pointer and is used to access the device tree.

Surprisingly, I found these two addresses are different.
Before entering the kernel, the pointer is: `0x80a00000`, and it turns to be `0x00000100`, which is related to the error adress `0x00000101`.

While if I initialized the value of the pointer to `0x80a00000` manually in `setup_vm` from [`arch/riscv/mm/init.c`](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/riscv/mm/init.c?id=d45331b00ddb179e291766617259261c112db872), the fdt_check_header will be executed successfully. And the information of deviced tree is same as the information listed after I add the option `--enable-print-device-tree`. However, set the Linux still do not print any words in console after I set the device tree pointer manually.

I do not know how to resolve this error. If there is any suggestion, please help me.

Thanks for your time.

Liboya-CB, 2019-09-20:
I update `riscv-isa-sim` and `riscv-gnu-toolchain` to the latest version. And the pointer adress of device tree after entering kernel is same as it is in bootloader.

1. [riscv-isa-sim](https://github.com/riscv/riscv-isa-sim/tree/a515af6d3bb5146a0a68cf88adf032fe78ba4ead)
2. [riscv-gnu-toolchain](https://github.com/riscv/riscv-gnu-toolchain/tree/e0283b704e919d5b507b50f459750115297c5b9b)

However, the kernel still does not print any words in console and is halted after scheduling several times.

Liboya-CB, 2019-09-23:
I checkout the branch `riscv-linux-4.20` from [riscv-linux](https://github.com/riscv/riscv-linux/tree/riscv-linux-4.20).
And I have already set these configuration:
```shell
CONFIG_HVC_RISCV_SBI=y

CONFIG_SERIAL_EARLYCON=y
CONFIG_SERIAL_8250=y
# CONFIG_SERIAL_8250_DEPRECATED_OPTIONS is not set
CONFIG_SERIAL_8250_FINTEK=y
CONFIG_SERIAL_8250_CONSOLE=y
CONFIG_SERIAL_8250_DMA=y
```

The Linux finally print these lines and hang.
```shell
bbl loader
 {
  #address-cells = <0x00000002>;
  #size-cells = <0x00000002>;
  compatible = "ucbbar,spike-bare-dev";
  model = "ucbbar,spike-bare";
  cpus {
    #address-cells = <0x00000001>;
    #size-cells = <0x00000000>;
    timebase-frequency = <0x00989680>;
    cpu@0 {
      device_type = "cpu";
      reg = <0x00000000>;
      status = "okay";
      compatible = "riscv";
      riscv,isa = "rv64imafdc";
      mmu-type = "riscv,sv48";
      clock-frequency = <0x3b9aca00>;
      interrupt-controller {
        #interrupt-cells = <0x00000001>;
        interrupt-controller;
        compatible = "riscv,cpu-intc";
        linux,phandle = <0x00000001>;
        phandle = <0x00000001>;
      }
    }
  }
  memory@80000000 {
    device_type = "memory";
    reg = <0x00000000 0x80000000 0x00000000 0x80000000>;
  }
  soc {
    #address-cells = <0x00000002>;
    #size-cells = <0x00000002>;
    compatible = "ucbbar,spike-bare-soc", "simple-bus";
    ranges;
  }
  htif {
    compatible = "ucb,htif0";
  }
}
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.20.0+ (liboya@M8600t) (gcc version 8.3.0 (GCC)) #18 SMP Mon Sep 23 09:49:51 CST 2019
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000fffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] software IO TLB: mapped [mem 0xf9ffe000-0xfdffe000] (64MB)
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 19 pages/cpu @(____ptrval____) s37720 r8192 d31912 u77824
[... 8212 lines cut ...]
# CONFIG_RCU_TRACE is not set
# CONFIG_RCU_EQS_DEBUG is not set
# CONFIG_DEBUG_WQ_FORCE_RR_CPU is not set
# CONFIG_DEBUG_BLOCK_EXT_DEVT is not set
CONFIG_NOTIFIER_ERROR_INJECTION=m
CONFIG_PM_NOTIFIER_ERROR_INJECT=m
# CONFIG_NETDEV_NOTIFIER_ERROR_INJECT is not set
# CONFIG_FAULT_INJECTION is not set
# CONFIG_LATENCYTOP is not set
CONFIG_HAVE_FUNCTION_TRACER=y
CONFIG_HAVE_FUNCTION_GRAPH_TRACER=y
CONFIG_HAVE_DYNAMIC_FTRACE=y
CONFIG_HAVE_DYNAMIC_FTRACE_WITH_REGS=y
CONFIG_HAVE_FTRACE_MCOUNT_RECORD=y
CONFIG_TRACING_SUPPORT=y
# CONFIG_FTRACE is not set
# CONFIG_DMA_API_DEBUG is not set
# CONFIG_RUNTIME_TESTING_MENU is not set
CONFIG_MEMTEST=y
# CONFIG_BUG_ON_DATA_CORRUPTION is not set
# CONFIG_SAMPLES is not set
# CONFIG_UBSAN is not set
CONFIG_EARLY_PRINTK=y
```
</details>

Liboya-CB, 2019-10-22:
After setting `CONFIG_VT_CONSOLE=n` while configuring linux by `make ARCH=riscv menuconfig`, it works smoothly. If CONFIG_VT_CONSOLE is enabled, a default console will be used instead of SBI.

Related configuration is found in /linux/drivers/tty/vt/vt.c

```C
static int __init con_init(void){
...
#ifdef CONFIG_VT_CONSOLE
	register_console(&vt_console_driver);
#endif
	return 0;
}
```

closed 2019-10-22 by Liboya-CB

RobertHenry6bev, 2019-10-27:
Do you have patchfiles, PRs, or branches I could use to try your approach to running on spike?

airobot60, 2020-08-11:
> Hi,
>
>
>
> I am trying to simulate latest Linux with spike on Ubuntu 16.04. But it seems that the device tree cannot be initialized successfully.
>
>
>
> The version of related repositories as listed as [...]