riscv-software-src/riscv-pk #184: machine/mtrap.c: determining absolute timeout from S-mode

niwis, 2020-01-10:
Hi!

When calling `SBI_SET_TIMER`, the passed argument is directly assigned to `mtimecmp`. Therefore, the argument is expected to be the absolute time the timer should trigger at. However, I cannot see how a supervisor is supposed to determine the absolute time without access to `mtime`. Wouldn't it make sense to expect a relative timeout instead and add `mtime` just before assigning to `mtimecmp` in `machine/mtrap.c:91`? This is also the behavior I understand from the [riscv-sbi-doc](https://github.com/riscv/riscv-sbi-doc/blob/master/riscv-sbi.adoc):

> ```
> void sbi_set_timer(uint64_t stime_value)
> ```
> Programs the clock for next event after *stime_value* time. This function also clears the pending timer interrupt bit.

Best,
Nils

aswaterman, 2020-01-10:
The supervisor can read the time CSR.

niwis, 2020-01-10:
Are the contents of `time` and `mtime` guaranteed to be equal then? Doesn't the spec allow for different implementations, e.g. by making `time` dependant on the system clock and driving `mtime` with the rtc?

aswaterman, 2020-01-10:
The spec is that sbi_set_timer is in terms of the time CSR. It’s an
implementation detail that the SBI timer exactly equals mtimecmp. But a
consequence of that implementation detail is that time must equal mtime.

In a virtualized environment, the SBI timer would probably be offset from
mtimecmp, hence the time CSR would be offset from mtime by the same amount.