riscv-software-src/riscv-pk #189: Provide PK binary releases

gsauthof, 2020-02-07:
For getting started with Spike it would be useful to have binary releases of PK under https://github.com/riscv/riscv-pk/releases.

Usecase: One already has the RISC-V Binutils and Spike installed. With a binary PK release one could immediately test some user-space assembly code in Spike, i.e. without having to compile a full RISC-V GNU toolchain that is required to compile the PK.

aswaterman, 2020-02-07:
I don’t have time to set this up, but it makes sense.

We’d want a release script that builds several variants (RV32 and RV64;
IMAC and GC). We can probably make the GCV and GC variants the same, since
we do not plan to support emulation of missing vector features.

czhaneva, 2021-05-04:
Is this script currently supported?