riscv-software-src/riscv-pk #191: error:When I execute the command "make"

qs132, 2020-03-02:
I encountered this problem when I build riscv-pk.

../machine/minit.c:164:9: error: array subscript 1 is outside array bounds of 'uint32_t[1]' {aka 'unsigned int[1]'} [-Werror=array-bounds]
  164 |     if (bswap(p_external_dtb[1]) < MAX_EXTERNAL_DTB_SIZE) {
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
../machine/minit.c:159:19: note: while referencing 'external_dtb'
  159 |   extern uint32_t external_dtb;
      |                   ^~~~~~~~~~~~
../machine/minit.c:189:7: error: array subscript 1 is outside array bounds of 'uint32_t[1]' {aka 'unsigned int[1]'} [-Werror=array-bounds]
  189 |       printm("The external dtb is too large(size = %d). Use default dtb.\n", bswap(p_external_dtb[1]));
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../machine/minit.c:159:19: note: while referencing 'external_dtb'
  159 |   extern uint32_t external_dtb;
      |                   ^~~~~~~~~~~~
cc1: all warnings being treated as errors
Makefile:322: recipe for target 'minit.o' failed

Hope to get everyone's help！
Tanks!

aswaterman, 2020-03-02:
I don't see any of this code on HEAD of master (`git grep MAX_EXTERNAL_DTB_SIZE` returns no hits).  Is this someone's custom branch?  (in which case you'll need to talk to whoever wrote that code)

qs132, 2020-03-03:
> I don't see any of this code on HEAD of master (`git grep MAX_EXTERNAL_DTB_SIZE` returns no hits). Is this someone's custom branch? (in which case you'll need to talk to whoever wrote that code)

Got it,tank you very much!

qs132, 2020-03-03:
> I don't see any of this code on HEAD of master (`git grep MAX_EXTERNAL_DTB_SIZE` returns no hits). Is this someone's custom branch? (in which case you'll need to talk to whoever wrote that code)

Got it,thank you very much!