riscv-software-src/riscv-pk #193: Illegal Instruction on linux on top BBL

johnnavaj, 2020-03-26:
I have built the mainline linux and passed it as a payload to BBL . Using minimal initramfs as filesystem. 

I am getting a illegal instruction once i drop into the userspace , what could be the cause of this trap ?
 
Log as below 

BusyBox v1.26.2 (2020-03-26 18:36:07 IST) built-in shell (ash)

# [   91.675392] Oops - illegal instruction [#1]
[   91.678436] CPU: 0 PID: 0 Comm: swapper Not tainted 5.5.0-rc1-dirty #3
[   91.683554] epc: ffffffe00007a3c0 ra : ffffffe0000a32c4 sp : ffffffe000289f50
[   91.689161]  gp : ffffffe0002a5d18 tp : ffffffe00028e428 t0 : 00000000000001f8
[   91.694838]  t1 : 0000000000000002 t2 : 0000000000000000 s0 : ffffffe000289f60
[   91.700519]  s1 : ffffffe0002a5aec a0 : 00000000000000cf a1 : ffffffe00028e428
[   91.706198]  a2 : ffffffe003831d60 a3 : 0000000000000000 a4 : 0000000000200002
[   91.711876]  a5 : 0000000000000000 a6 : 00000000149d72a0 a7 : ffffffe003200038
[   91.717557]  s2 : ffffffe007c3de80 s3 : ffffffe0002a5830 s4 : 0000000000000001
[   91.723236]  s5 : ffffffe0002a5838 s6 : ffffffe000015018 s7 : 0000000000000000
[   91.728911]  s8 : 0000000000000000 s9 : 0000000000000000 s10: 0000000000000000
[   91.734593]  s11: 0000000000000000 t3 : 0000000000000402 t4 : ffffffe000294168
[   91.740266]  t5 : ffffffe000294170 t6 : 0000000000000040
[   91.744448] status: 0000000200000100 badaddr: 0000000010500073 cause: 0000000000000002
[   91.750672] ---[ end trace c87edc2fc8b8df75 ]---
[   91.754310] Kernel panic - not syncing: Attempted to kill the idle task!
[   91.759577] ---[ end Kernel panic - not syncing: Attempted to kill the idle task! ]---

sahajgarg, 2020-08-18:
What emulator/simulator are you using? 
If you are using TinyEMU, try removing COUNTEREN_MASK from riscv_cpu.c. This disables many counters in user mode (bbl in riscv-pk appears to allow all counters in machine mode, regardless of whether the counters are disabled, which is why the issue only appears for user mode).

sahajgarg, 2020-08-18:
The badaddr field (mtval) contains the faulting instruction, which appears to be a read to the CSR 0x105, which corresponds to stvec. This is supervisor read/write privilege, so that might be causing the illegal instruction.

closed 2021-01-13 by johnnavaj