riscv-software-src/riscv-pk #195: Run Newlib ELFs on FPGA

genesys2, 2020-03-31:
Hi,

I am trying to run benchmarks on a vexriscv CPU (rv32ima) and I need them to run on an FPGA board (Genesys2). Running Linux to perform the benchmarks is not an option as I did some hardware and compiler modifications and I cannot compile the Linux kernel with LLVM. The whole LLVM thing is another problem, so for now I am just going to use gcc.

So, my plan is to run the benchmarks without OS on the FPGA. Therefore, I installed the riscv32-unknown-elf toolchain and the proxy kernel. Unfortunately, i cannot find any instructions on how to use the pk without spike. 

I am using LiteX to load the binary files to the FPGA - if I do 
`./litex_term.py --speed 1e6 --serial-boot --kernel pk.bin --no-crc /dev/ttyUSB0`, LiteX gives the following output (It doesn't crash but doesn't do anything either so I'm not 100% sure whether this is correct)
````
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading pk.bin to 0x40000000 (66652 bytes)...
[LXTERM] Upload complete (80.9KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
CCC
````

Obviously, I still need to load my executable somewhere but I don't know where. I loaded the pk elf into gdb but didn't find anything useful. LiteX does support loading multiple files to different addresses but I don't know which file to load where.

The "CCC" bit in the output above is random and does not appear everytime which I think might be a sign that the pk does it's thing and then jumps to some address where nothing is (?)

Now, I am not sure whether what it did until now is correct or if there is a better way to run executables without OS on the FPGA. So if someone knows this or has an idea, I would be very happy if you could help me out.

Thanks!

tommythorn, 2020-03-31:
pk depends heavily of the htif interface to communicate with a host-side server, sort of the proxy for all the system call work.  Once loaded, _it_ loads your binaries using proxy services.  I know nothing of LiteX but I would be surprised if litex_term implements the services and transport required (which involves it magically reading and writing memory in the process).

To get this to work, you'd need to implement the same transport and bridge the communication.  TBH, this is challenging as the whole thing is essentially undocumented and riddled with assumptions.

genesys2, 2020-04-01:
I was kinda expecting something like this so I might have to look at other options to run my code. 

Anyway, thanks for your help and the quick response! :-)

closed 2020-04-01 by genesys2

tommythorn, 2020-04-01:
De nada.  I have had a similar desire and gone through some of the motions but I’ve so far bailed on pk.