riscv-software-src/riscv-pk #196: Use proxy kernel in co-simulation gets stuck in boot_other_hart

Marc43, 2020-03-31:
Hi everyone,

I am trying to use the proxy kernel together with spike to run my C applications in co-simulation to feed a processor. In the process of bringing up this structure now I am getting proxy kernel inside spike stucked here:

```
00000000800006ba <boot_other_hart>:                                                                  
    800006ba:   10500073            wfi                                                              
    800006be:   bff5                    j   800006ba <boot_other_hart>  
```

As there is no documentation I don't know what this does, do you have any pointer on what this function does, and how should I continue? 

Any help is appreciated.

Thanks,
Marc

aswaterman, 2020-04-01:
PK only supports 1 hart (with hart id 0).  The other harts just spin forever in that loop.

Marc43, 2020-04-01:
Didn't know that, but neither I tried to do so. Do you have any idea of how I could have attempted to start another hart? It's a fresh pull of the pk repository using spike.

aswaterman, 2020-04-01:
No idea. I think you’d know if you had multiple harts.

closed 2020-04-07 by Marc43