riscv-software-src/riscv-pk #197: Operation in redirect_trap

Phantom1003, 2020-04-08:
https://github.com/riscv/riscv-pk/blob/8c125897999720856262f941396a9004b0ff5d3d/machine/mtrap.c#L183
https://github.com/riscv/riscv-pk/blob/8c125897999720856262f941396a9004b0ff5d3d/machine/mtrap.c#L184
code in redirect_trap makes me really confuse:
```c
void redirect_trap(uintptr_t epc, uintptr_t mstatus, uintptr_t badaddr)
{
  // set mepc to s mode handler
  // when mret switch to s mode to handle (redirect trap)
  write_csr(sbadaddr, badaddr);
  write_csr(sepc, epc);
  write_csr(scause, read_csr(mcause));
  write_csr(mepc, read_csr(stvec));
  // set S previous privilege to U and disable s mode interrupt
  uintptr_t new_mstatus = mstatus & ~(MSTATUS_SPP | MSTATUS_SPIE | MSTATUS_SIE);
  // set M previous mode from M to S
  uintptr_t mpp_s = MSTATUS_MPP & (MSTATUS_MPP >> 1);
  // x * (MSTATUS_SPIE / MSTATUS_SIE) == x << 4
  // xPIE <- xIE
  @1：new_mstatus |= (mstatus * (MSTATUS_SPIE / MSTATUS_SIE)) & MSTATUS_SPIE;
  // (mpp_s / MSTATUS_SPP) = 1c
  @2：new_mstatus |= (mstatus / (mpp_s / MSTATUS_SPP)) & MSTATUS_SPP;
  new_mstatus |= mpp_s;
  write_csr(mstatus, new_mstatus);
  extern void __redirect_trap();
  return __redirect_trap();
}
```
I don't think operation at @1 and @2 are valid.
@1: not all field are aligned 4, for MPP and other field, directly shift 4 bytes will makes dislocation
@2: how to unserstand / 1c ...

Thanks

aswaterman, 2020-04-09:
If you look at the generated assembly code you'll see it does the right thing.

```
slli    a5,a1,0x4    # Move SIE up to SPIE
andi    a5,a5,32     # Isolate new SIE
srli    a4,a1,0x3    # Move MPP[0] down to SPP
andi    a4,a4,256    # Isolate new SPP
or      a5,a5,a4     # OR the new SPP and SPIE together
lui     a4,0x1
addi    a4,a4,-2048
or      a5,a5,a4     # MPP = S
```

Phantom1003, 2020-04-12:
wait ...

closed 2020-04-12 by Phantom1003