riscv-software-src/riscv-pk #198: Error compiling pk

CC-Hi, 2020-04-12:
I have installed riscv-gnu-toolchain and riscv-tools, but when installing riscv-tools, riscv-pk was not successfully installed, so I installed riscv-pk separately, and the following error occurred during compilation:
`gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32i -mabi=ilp32 -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/file.c
gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32i -mabi=ilp32 -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/syscall.c
gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32i -mabi=ilp32 -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/handlers.c
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
gcc: error: unrecognized argument in option ‘-mabi=ilp32’
gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32i -mabi=ilp32 -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/frontend.c
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32i -mabi=ilp32 -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/elf.c
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
gcc: error: unrecognized argument in option ‘-mabi=ilp32’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
gcc: error: unrecognized argument in option ‘-mabi=ilp32’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
Makefile:319: recipe for target 'file.o' failed
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
make: *** [file.o] Error 1
gcc: error: unrecognized argument in option ‘-mabi=ilp32’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
make: *** waiting for unfinished task....
Makefile:319: recipe for target 'handlers.o' failed
make: *** [handlers.o] Error 1
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
gcc: error: unrecognized argument in option ‘-mabi=ilp32’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
Makefile:319: recipe for target 'syscall.o' failed
make: *** [syscall.o] Error 1
Makefile:319: recipe for target 'frontend.o' failed
make: *** [frontend.o] Error 1
Makefile:319: recipe for target 'elf.o' failed
make: *** [elf.o] Error 1

`
Thank you for helping me!

Kelvinyu1117, 2020-04-12:
same problem
I have checked my PATH should be correct
RISCV = /opt/riscv
PATH = $RISCV/bin:$PATH

CC-Hi, 2020-04-12:
> same problem
> I have checked my PATH should be correct
> RISCV = /opt/riscv
> PATH = $RISCV/bin:$PATH

@Kelvinyu1117   Yeh,my PATH is the same with yours. I found that the command

> ./build.sh
may cause the error. I successfully compiled a PK executable file with ./build-rv32ima.sh this command, but it still cannot be used.I am now trying to recompile.

CC-Hi, 2020-04-12:
> same problem
> I have checked my PATH should be correct
> RISCV = /opt/riscv
> PATH = $RISCV/bin:$PATH
@Kelvinyu1117 
Yeh,my PATH is the same with yours. I found that the command “./build.sh” may cause this error. I successfully compiled a PK executable file with ./build-rv32ima.sh this command, but it still cannot be used.I am now trying to recompile.

Kelvinyu1117, 2020-04-12:
@CC-Hi  I see, can you tell me the solution if you figure out it? Thank you very much

CC-Hi, 2020-04-12:
> @CC-Hi I see, can you tell me the solution if you figure out it? Thank you very much

@Kelvinyu1117  Still no success.I don't know why the PK was successfully compiled before.
Here is the message:
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
gcc: error: unrecognized argument in option ‘-mabi=ilp32’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
make: *** [file.o] Error 1

Kelvinyu1117, 2020-04-12:
@CC-Hi That's my error message actually
I can use riscv64-unknown-elf-gcc to compile my helloworld program
but still can't compile the PK

CC-Hi, 2020-04-14:
> @CC-Hi That's my error message actually
> I can use riscv64-unknown-elf-gcc to compile my helloworld program
> but still can't compile the PK

@Kelvinyu1117  I have installed the rv64 pk successfully. You can try to use './build-spike-pk.sh' to building your riscv-tools.
But I still can't install the rv32 pk.(of course I have installed the rv32 riscv-gnu-toolchain when I build the rv32 pk).

Kelvinyu1117, 2020-04-14:
@CC-Hi 
I have also get it done.
I think the trick is that when we build the riscv-gnu-toolchain,
we shouldn't run the ./configure with the prefix.
Then you just need to follow the instructions stated in the repo of riscv-tools