riscv-software-src/riscv-pk #201: Questions about building pk

WilliamWangPeng, 2020-05-21:
Hi guys
    My honor to ask questions here, recently I'm building pk, and I follow the steps in README.md
$ mkdir build
$ cd build
$ ../configure --prefix=$RISCV --host=riscv64-unknown-elf
$ make
$ make install

after $ ../configure --prefix=/home/u/tools --host=riscv64-unknown-elf 
there remain some errors as following, anyone knows how to correct? (Ubuntu 18.04 )
gcc: error: unrecognized argument in option ‘-mcmodel =medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
make: *** [file.o] Error 1

aswaterman, 2020-05-21:
That's an error that occurs when your host x86 compiler gets used in place of your RISC-V cross-compiler.  Make sure riscv64-unknown-elf-gcc is in your PATH when you run configure.

WilliamWangPeng, 2020-05-22:
> That's an error that occurs when your host x86 compiler gets used in place of your RISC-V cross-compiler. Make sure riscv64-unknown-elf-gcc is in your PATH when you run configure.

thank you Andrew Waterman, the problem has solved

Fifiy, 2020-06-26:
@WilliamWangPeng Can you please tell me how did you solve it ?

closed 2020-08-03 by WilliamWangPeng