riscv-software-src/riscv-pk #204: make error : gcc: error: unrecognized argument in option ‘-mcmodel=medany’

mikemusasa, 2020-06-29:
Hi, i 've cloned all of the riscv tools and cant seem to  make the make command of the riscv-pk work. i keep receiving this error : 

gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
make: *** [Makefile:332: file.o] Error 1  

any idea on how to fix it ?  
 i already have the riscv toolchain installed  as well ) and running this on ubuntu, trying to use the riscv  rocket processor

aswaterman, 2020-06-30:
This means you’re using a host compiler rather than a RISC-V compiler. Make sure that riscv64-unknown-elf-gcc is in your PATH when you execute configure

Fifiy, 2020-07-03:
I have the same issue, you have to check your path, make sure you can reach riscv64-unknown-elf-gcc because that's what it wants.

Fifiy, 2020-07-03:
I had the same issue, you have to check your path, make sure you can reach riscv64-unknown-elf-gcc because that's what it wants.

Fifiy, 2020-07-03:
I had the same issue, you have to check your path, make sure you can reach riscv64-unknown-elf-gcc because that's what it wants. Make sure riscv64-unknown-elf-gcc is in your PATH when you run configure.

mikemusasa, 2020-07-03:
thanks for the answers,  i am pretty sure it should be able to reach it. 
In my bashrc file i added these two lines : 
 export RISCV=/opt/riscv
export PATH=$RISCV/bin:$PATH  
 
and when i run the whereis  riscv64-unknown-elf-gcc,  the response is  /opt/riscv/bin/riscv64-unknown-elf-gcc. 
So it should be able to reach it, since it's just in the inside bin which is part of my path or did i missunderstood something ? 
once again thanks for the help

scottj97, 2020-07-12:
@mikemusasa make sure you're rerun `../configure` with this in your PATH. Delete the entire `build` directory as a last resort.

mikemusasa, 2020-07-13:
I deleted the build and started over again and i noticed when running configure , this is the messages that the configuration prints out i s below. If i udnerstand it correctly, it seems that it  checks for a different riscv-unknown-linux-gnu things that it's can not find ? 

below is the configure message : 

checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-linux-gnu
checking for riscv64-unknown-linux-gnu-gcc... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for riscv64-unknown-linux-gnu-g++... no
checking for riscv64-unknown-linux-gnu-c++... no
checking for riscv64-unknown-linux-gnu-gpp... no
checking for riscv64-unknown-linux-gnu-aCC... no
checking for riscv64-unknown-linux-gnu-CC... no
checking for riscv64-unknown-linux-gnu-cxx... no
checking for riscv64-unknown-linux-gnu-cc++... no
checking for riscv64-unknown-linux-gnu-cl.exe... no
checking for riscv64-unknown-linux-gnu-FCC... no
checking for riscv64-unknown-linux-gnu-KCC... no
checking for riscv64-unknown-linux-gnu-RCC... no
checking for riscv64-unknown-linux-gnu-xlC_r... no
checking for riscv64-unknown-linux-gnu-xlC... no
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for riscv64-unknown-linux-gnu-ar... no
checking for ar... ar
checking for riscv64-unknown-linux-gnu-ranlib... no
checking for ranlib... ranlib
checking for riscv64-unknown-linux-gnu-readelf... no
checking for readelf... readelf
checking for riscv64-unknown-linux-gnu-objcopy... no
checking for objcopy... objcopy
checking for a BSD-compatible install... /usr/bin/install -c
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
configure: configuring default subproject : pk
configure: configuring default subproject : bbl
configure: configuring default subproject : softfloat
configure: configuring default subproject : dummy_payload
configure: configuring default subproject : machine
configure: configuring default subproject : util
configure: creating ./config.status
config.status: creating pk.mk
config.status: creating bbl.mk
config.status: creating softfloat.mk
config.status: creating dummy_payload.mk
config.status: creating machine.mk
config.status: creating util.mk
config.status: creating Makefile
config.status: creating config.h

scottj97, 2020-07-13:
Are you following the build steps in the README? The `--host` option to `configure` should handle this.

```
$ ../configure --prefix=$RISCV --host=riscv64-unknown-elf
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-elf
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether riscv64-unknown-elf-gcc accepts -g... yes
checking for riscv64-unknown-elf-gcc option to accept ISO C89... none needed
checking for riscv64-unknown-elf-g++... riscv64-unknown-elf-g++
checking whether we are using the GNU C++ compiler... yes
checking whether riscv64-unknown-elf-g++ accepts -g... yes
checking for riscv64-unknown-elf-ar... riscv64-unknown-elf-ar
checking for riscv64-unknown-elf-ranlib... riscv64-unknown-elf-ranlib
checking for riscv64-unknown-elf-readelf... riscv64-unknown-elf-readelf
checking for riscv64-unknown-elf-objcopy... riscv64-unknown-elf-objcopy
checking for a BSD-compatible install... /usr/bin/install -c
checking how to run the C preprocessor... riscv64-unknown-elf-gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
configure: configuring default subproject : pk
configure: configuring default subproject : bbl
configure: configuring default subproject : softfloat
configure: configuring default subproject : dummy_payload
configure: configuring default subproject : machine
configure: configuring default subproject : util
configure: creating ./config.status
config.status: creating pk.mk
config.status: creating bbl.mk
config.status: creating softfloat.mk
config.status: creating dummy_payload.mk
config.status: creating machine.mk
config.status: creating util.mk
config.status: creating Makefile
config.status: creating config.h
```

mikemusasa, 2020-07-13:
yes, i followed all of the build steps from : https://github.com/riscv/riscv-pk/

YFJiang, 2020-07-18:
Hi mikemusasa:
You can check what your real host argument is, take my pc for example:
johnson@johnson-ubuntu:~/riscv/riscv-tools/tmp$ which riscv64-unknown-linux-gnu-gcc
/opt/riscv/bin/riscv64-unknown-linux-gnu-gcc

So my configure should be:
../configure --prefix=$RISCV --host=riscv64-unknown-linux-gnu
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-linux-gnu
checking for riscv64-unknown-linux-gnu-gcc... riscv64-unknown-linux-gnu-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether riscv64-unknown-linux-gnu-gcc accepts -g... yes
checking for riscv64-unknown-linux-gnu-gcc option to accept ISO C89... none needed
checking for riscv64-unknown-linux-gnu-g++... riscv64-unknown-linux-gnu-g++
checking whether we are using the GNU C++ compiler... yes
checking whether riscv64-unknown-linux-gnu-g++ accepts -g... yes
checking for riscv64-unknown-linux-gnu-ar... riscv64-unknown-linux-gnu-ar
checking for riscv64-unknown-linux-gnu-ranlib... riscv64-unknown-linux-gnu-ranlib
checking for riscv64-unknown-linux-gnu-readelf... riscv64-unknown-linux-gnu-readelf
checking for riscv64-unknown-linux-gnu-objcopy... riscv64-unknown-linux-gnu-objcopy
checking for a BSD-compatible install... /usr/bin/install -c
checking how to run the C preprocessor... riscv64-unknown-linux-gnu-gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
configure: configuring default subproject : pk
configure: configuring default subproject : bbl
configure: configuring default subproject : softfloat
configure: configuring default subproject : dummy_payload
configure: configuring default subproject : machine
configure: configuring default subproject : util
configure: creating ./config.status
config.status: creating pk.mk
config.status: creating bbl.mk
config.status: creating softfloat.mk
config.status: creating dummy_payload.mk
config.status: creating machine.mk
config.status: creating util.mk
config.status: creating Makefile
config.status: creating config.h
config.status: config.h is unchanged

Make sure no space before and after '=' in the host option.

BR,
johnson

Cedricyc, 2020-08-18:
Hi there, I met the exactly same question as yours.
The main problem is that you set the $PATH but in ./configure the $PATH doesn't work.
It may be the problem of sudo
IF you use sudo ./configure --prefix=blahblahblah --host=blahblahbah
sudo's $PATH is not equal to your user $PATH, you need to add riscv toolchain path into sudo $PATH

one way to do this is let sudo's PATH derive from user PATH 
(1) sudo visudo, change Default_env_reset into !Default_env_reset
(2)change ~/.bashrc, append one line at end of the file: 
alias sudo='sudo env PATH=$PATH'

Hope it helps : )

Ytm0316, 2020-10-11:
@Cedricyc  Hello, your answer helps me a lot, thank you!
And I  use      
     $sudo env "PATH=$PATH" ../configure --prefix=$RISCV --host=riscv64-unknown-elf
It enables the program to find where my cross compiler is installed in.

theverytak, 2021-01-15:
I believe we can close this issue if the threads above helped you out @mikemusasa

mikemusasa, 2021-01-18:
Yes , forgot to close it .

closed 2021-01-18 by mikemusasa

Jingqing3948, 2023-11-11:
> 嗨，我遇到了和你完全一样的问题。主要问题是您设置了$PATH，但在 ./configure 中$PATH不起作用。可能是sudo的问题 如果你使用 sudo ./configure --prefix=blahblahblah --host=blahblahbah sudo 的$PATH不等于你的用户$PATH，你需要在 sudo $PATH中添加 riscv 工具链路径
> 
> 一种方法是让 sudo 的 PATH 派生自用户 PATH （1） sudo visudo，Default_env_reset改成！Default_env_reset （2）更改 ~/.bashrc，在文件末尾附加一行：alias sudo='sudo env PATH=$PATH'
> 
> 希望它有所帮助：）cyc

Thanks So much π_π