riscv-software-src/riscv-pk #207: How can I use spike and pk to run a dynamic linked elf.

ALLap97, 2020-07-13:
I use riscv64-unknown-linux-gnu-gcc/g++ to  compile opencv and get the dynamic library of opencv.
I can run the test using qemu_riscv64 successfully.
`
qemu-riscv64 -L /opt/riscv/64/sysroot DisplayImage
`

but when i use spike and pk to run the test , it occurs some errors.
`
spike pk DisplayImage
`

error
`
bbl loader
z  0000000000000000 ra 0000000000011504 sp 000000007f7e9b50 gp 0000000000018800
tp 0000000000000000 t0 0000000000000000 t1 0000000000000000 t2 0000000000018310
s0 0000000000000000 s1 0000000000000000 a0 00000000000115c8 a1 0000000000000001
a2 000000007f7e9b58 a3 0000000000013392 a4 00000000000133ea a5 0000000000000000
a6 000000007f7e9b50 a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 ffffffffffffffff t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc fffffffffffffffe va fffffffffffffffe insn       ffffffff sr 8000000200046020
User fetch segfault @ 0xfffffffffffffffe

`

i think the reason is that pk doesn't support danamic linked elf.
so how can i  I use spike and pk to run a dynamic linked elf.

aswaterman, 2020-07-13:
pk only supports statically linked ELFs.

In theory, you could invoke the dynamic linker directly, then pass the program as an argument (i.e. `spike pk ./ld.so.1 ./a.out`), but there's almost certainly a lot of hacking you'll need to do to make that work.

You're better off doing a static link, or running a real OS like Linux instead of using the pk.

ALLap97, 2020-07-13:
> pk only supports statically linked ELFs.
>
> In theory, you could invoke the dynamic linker directly, then pass the program as an argument (i.e. `spike pk ./ld.so.1 ./a.out`), but there's almost [...]

ALLap97, 2020-07-13:
thinks for your advise.
but actually we also tried the static link。
>
`
riscv64-unknown-linux-gnu-g++  Dp.cpp    -o Dp  -static  -I /opt/opencv/static/include/opencv4/  -L /opt/opencv/static/lib -lopencv_imgcodecs -lopencv_imgproc  -lopencv_core -ldl  -L /opt/opencv/static/lib/opencv4/3rdparty  -llibwebp  -llibpng  -llibtiff  -llibjasper  -llibjpeg-turbo  -lquirc -lade  -lzlib -lpthread
`
>
we can get the elf with a waring.
`
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /opt/opencv/static/lib/libopencv_core.a(opencl_core.cpp.o): in function `.L17':
opencl_core.cpp:(.text._ZL15opencl_check_fni+0x19a): warning: Using 'dlopen' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking

`
>
we can run the elf with qemu_riscv64.
but we got a error when with spike and pk
`
spike pk Dp
`
>
error
`
bbl loader

bad syscall #98!
`
do you think the waring is the reason?
**warning: Using 'dlopen' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking**
how can we fix it?

aswaterman, 2020-07-13:
Probably unrelated.  Syscall 98 is `futex`, which `pk` doesn't support.  (`pk` supports only a tiny subset of syscalls.)  Using dynamic linking would not help.

ALLap97, 2020-07-13:
thinks again.
we will try the real OS(linux) way.

closed 2020-07-13 by ALLap97

czhaneva, 2021-05-04:
> thinks for your advise.
> but actually we also tried the static link。
>
>
> `riscv64-unknown-linux-gnu-g++ Dp.cpp -o Dp -static -I /opt/opencv/static/include/opencv4/ -L /opt/opencv/static/lib [...]

I recently met the same problem, I cross compiled the opencv repo for my project, when I tested it, I got bas syscall #98!
So I'd like to know whether you fixed it? and how to fix it~
Thanks

ALLap97, 2021-05-04:
> I recently met the same problem, I cross compiled the opencv repo for my project, when I tested it, I got bas syscall #98!
> So I'd like to know whether you fixed it? and how to fix it~
> Thanks

We modified the pk code, we added an empty syscall (num 98) in pk. And it can run normally eventually.

czhaneva, 2021-05-04:
> We modified the pk code, we added an empty syscall (num 98) in pk. And it can run normally eventually.

Thanks for your quick reply, could you please describe in detail what should I do?
My understanding is I need to define it in syscall.h as following:
`#define SYS_98call 98`
and add an empty function in <riscv-pk/pk.syscall.c> as following:
`void syscall_98() {}`
and add it in do_syscall function as following
`[SYS_98call ] = syscall_98,`

I look forward to hearing from you.

ALLap97, 2021-05-04:
> Thanks for your quick reply, could you please describe in detail what should I do?
> My understanding is I need to define it in syscall.h as following:
> `#define SYS_98call 98`
> and add an empty [...]

That's right, this is the same as what I did.
```
#define SYS_futex = 98
...
static int sys_futex(){
    return 0;
}
...
[SYS_futex] = sys_futex,
```

czhaneva, 2021-05-04:
> That's right, this is the same as what I did.
>
> ```
> #define SYS_futex = 98
> ...
> static int sys_futex(){
>     return 0;
> }
> ...
> [SYS_futex] = sys_futex,
> ```

Thank you very much! It helps me a lot!