riscv-software-src/riscv-pk #212: pk：cannot execute binary file: Exec format error

BeenLi, 2020-08-17:
I have installed riscv64-unknown-linux-gnu pk succesfully in my utunbu at x86_64，but when I enter the commad `pk`
The following error occurred.

> oneshot@ubuntu:~/riscv/riscv-pk/build$ pk
> bash: /opt/riscv/riscv64-unknown-linux-gnu/bin/pk: cannot execute binary file: Exec format error

Any help is appreciated.

kito-cheng, 2020-08-17:
You need run pk with spike / [riscv-isa-sim](https://github.com/riscv/riscv-isa-sim).
e.g. 
```
$ spike pk <your-riscv-program>
```

BeenLi, 2020-08-19:
Thanks.
But I still can't run the hello successfully using your method.

Can you help me?

 

 $ spike pk <your-riscv-program&gt;

BeenLi, 2020-08-25:
@kito-cheng Thanks a lot, But I'm poor about how to build hello program using static link. Can you talk about the whole process in detail.

BeenLi, 2020-08-25:
@seldridge Thanks. But it seems that it doesn't work in your way.

gjunyi, 2021-12-07:
just like
$ riscv64-unknown-linux-gnu-gcc -static hello.c

jundijiujieke, 2023-05-04:
[image]
bbl loader
z  0000000000000000 ra ffffffc0000041dc sp ffffffc000417d30 gp 0000000000000000
tp 0000000000000000 t0 0000000000000025 t1 ffffffc000417d6f t2 0000000000000000
s0 ffffffc000417d31 s1 ffffffc0000130a8 a0 000000000000002e a1 0000000000000100
a2 ffffffc00000d394 a3 0000000010000000 a4 0000000010000005 a5 0000000000000000
a6 000000000000003f a7 000000000000003f s2 ffffffc0000130b8 s3 ffffffc0000130c0
s4 ffffffc00000f018 s5 0000000000000001 s6 ffffffc000010478 s7 ffffffc000013020
s8 ffffffc000010480 s9 ffffffc000010488 sA 0000000000000001 sB 0000000000000000
t3 fffffffffffffffc t4 ffffffc00000d140 t5 0000000000000009 t6 0000000000000015
pc ffffffc0000052d4 va/inst 0000000010000005 sr 8000000200006100
Kernel load segfault @ 0x0000000010000005
when i run spike pk hello. It doesn't print "hello world", and its result likes this.
Dose it work sucessfully?
And when i check the pk version. It prints "riscv64-unknown-elf/bin/pk: cannot execute binary file: Exec format error"

jundijiujieke, 2023-05-04:
[image]
when i run spike pk hello. It doesn't print "hello world", and its result likes this.
Dose it work sucessfully?
And when i check the pk version. It prints "riscv64-unknown-elf/bin/pk: cannot execute binary file: Exec format error"