riscv-software-src/riscv-pk #214: delegate_traps assertion failed when spike supports H extension

syvenwang7, 2020-08-20:
When spike supports H extension, the some bits of  mideleg are tied to 1.
The assertions in delegate_traps function are failed.

aswaterman, 2020-08-20:
I think pk needs to clear the misa.H bit.

cc @avpatel - in Spike, does clearing the misa.H bit get rid of the mideleg bits that are tied to 1

avpatel, 2020-08-20:
Spike does not clear the VS-interrupt bits tied to 1 in mideleg when misa.H bit is updated. We have to fix Spike for this.

Regards,
Anup

aswaterman, 2020-08-20:
Yeah, I think that is a minor bug that needs to get fixed.  In the mean time, you can work around the bug by commenting out the assertion, I hope.  (But we shouldn't upstream the workaround.)

avpatel, 2020-09-01:
I have send PR to Spike repo. Link https://github.com/riscv/riscv-isa-sim/pull/537/