riscv-software-src/riscv-pk #215: What does *trapframe->insn mean?

SKTT1Ryze, 2020-09-10:
This is a struct defined in pk.h:  
```
typedef struct
{
  long gpr[32];
  long status;
  long epc;
  long badvaddr;
  long cause;
  long insn;
} trapframe_t;
```

I don't understand what does ` insn ` member mean.  
And what does it work for or why we need this member?  
Anybody can give me some tips?  
Thank you very much.

wangnengjie, 2020-09-12:
I have same puzzle. I gusse it is "instruction name" which is the machine code of instructions.
Maybe this can help you:
https://github.com/riscv/riscv-isa-sim/blob/58f23e111fcfe317f366e2a77c04bad274f436d2/riscv/decode.h#L75

wangnengjie, 2020-09-12:
I think it is the "instruction name". It will store the machine code of the instruction. The machine code of instruction can be got by trapframe->epc, as it point to the addr of the instruction which cause interrupt.
Due to the specification of riscv instruction, bits have different meanings. So we need it.
maybe this can help you: https://github.com/riscv/riscv-isa-sim/blob/58f23e111fcfe317f366e2a77c04bad274f436d2/riscv/decode.h#L75

sorear, 2020-09-12:
pk is pretty small.  The definition:

https://github.com/riscv/riscv-pk/blob/master/pk/pk.h#L20

We're looking for something at an offset of 36 words.

`trapframe_t` is received by the `handle_trap` function:

https://github.com/riscv/riscv-pk/blob/master/pk/handlers.c#L101

which is called from assembly code:

https://github.com/riscv/riscv-pk/blob/master/pk/entry.S#L67

The trap frame is in `a0` (standard calling convention), which is immediately assigned from `sp`.  None of the normal instructions there write to `sp`, so let's look at the macro.

Aha! We have a write at an offset of 36 words.

https://github.com/riscv/riscv-pk/blob/master/pk/entry.S#L53

So `insn` is most likely completely unused.  What does the history say?  When was the `li x5, -1` added?  We check blame:

https://github.com/riscv/riscv-pk/blame/master/pk/entry.S#L52

which links to:

https://github.com/riscv/riscv-pk/commit/186ae3cc353fa3115d4ee0c9fcb18d8b2370c68c

unfortunately you do have to re-find the li x5 line at this point, but it's not so hard:

https://github.com/riscv/riscv-pk/commit/186ae3cc353fa3115d4ee0c9fcb18d8b2370c68c#diff-7cee668fe70986b23a70235ebd1d04ffR63

So this is a feature that was removed well before risc-v 2.0 back when there was a `mfpcr` instruction, and you can probably ignore it.

SKTT1Ryze, 2020-09-13:
> I think it is the "instruction name". It will store the machine code of the instruction. The machine code of instruction can be got by trapframe->epc, as it point to the addr of the instruction which cause interrupt.
> Due to the specification of riscv instruction, bits have different meanings.
> maybe this can help you: https://github.com/riscv/riscv-isa-sim/blob/58f23e111fcfe317f366e2a77c04bad274f436d2/riscv/decode.h#L75

thanks very much.

SKTT1Ryze, 2020-09-13:
> pk is pretty small. The definition:
> 
> https://github.com/riscv/riscv-pk/blob/master/pk/pk.h#L20
> 
> We're looking for something at an offset of 36 words.
> 
> `trapframe_t` is received by the `handle_trap` function:
> 
> https://github.com/riscv/riscv-pk/blob/master/pk/handlers.c#L101
> 
> which is called from assembly code:
> 
> https://github.com/riscv/riscv-pk/blob/master/pk/entry.S#L67
> 
> The trap frame is in `a0` (standard calling convention), which is immediately assigned from `sp`. None of the normal instructions there write to `sp`, so let's look at the macro.
> 
> Aha! We have a write at an offset of 36 words.
> 
> https://github.com/riscv/riscv-pk/blob/master/pk/entry.S#L53
> 
> So `insn` is most likely completely unused. What does the history say? When was the `li x5, -1` added? We check blame:
> 
> https://github.com/riscv/riscv-pk/blame/master/pk/entry.S#L52
> 
> which links to:
> 
> [186ae3c](https://github.com/riscv/riscv-pk/commit/186ae3cc353fa3115d4ee0c9fcb18d8b2370c68c)
> 
> unfortunately you do have to re-find the li x5 line at this point, but it's not so hard:
> 
> [186ae3c#diff-7cee668fe70986b23a70235ebd1d04ffR63](https://github.com/riscv/riscv-pk/commit/186ae3cc353fa3115d4ee0c9fcb18d8b2370c68c#diff-7cee668fe70986b23a70235ebd1d04ffR63)
> 
> So this is a feature that was removed well before risc-v 2.0 back when there was a `mfpcr` instruction, and you can probably ignore it.

thanks very much

SKTT1Ryze, 2020-09-15:
Maybe my problem has been solved, so close this issure.

closed 2020-09-15 by SKTT1Ryze