riscv-software-src/riscv-pk #218: Building bbl fails attempting to compile non-existent DTS file

repnop, 2020-10-30:
Looks like a161e6f3ef31004e47a5b94b85c2e84b764f8637 broke the build process. I don't provide any DTS files for it to use and spent the last while trying to understand why the build broke (I did some reorganization in a project that uses bbl and had the build silenced at first), it spits out
```
dtc -O dtb  -o machine.dtb
Error: <stdin>:0.0 syntax error
FATAL ERROR: Unable to parse input tree
make: *** [machine.mk:44: machine.dtb] Error 1
```
which stems from
https://github.com/riscv/riscv-pk/blob/a161e6f3ef31004e47a5b94b85c2e84b764f8637/machine/machine.mk.in#L43-L44
rolling back to the parent commit works

aswaterman, 2020-10-30:
Ugh.  Thanks.

closed 2020-10-30 by aswaterman

repnop, 2020-10-30:
Wow that was quick! Thank you! :)

commit def2f01e (2020-10-30): Revert "Disable device tree filter when load a dts from file (#217)"

This reverts commit a161e6f3ef31004e47a5b94b85c2e84b764f8637.

Resolves #218