riscv-software-src/riscv-pk #223: Failed boot from DDR which‘s address is larger than 2^32

yzt000000, 2020-11-22:
I had a version work well when the DDR address is 0x80000000, when I change the address to be 0x100000000 which is larger than 2^32.  BBL will always run into mtrap , then go into "poweroff".

does current BBL support larger 2^32 address?

aswaterman, 2020-11-24:
I believe it only uses the lowermost 2 GiB of physical memory, but moving the base to 0x100000000 works fine for me (either with or without an apparent bug fix I just made).  I configured pk with `--with-mem-start=0x100000000` and ran spike with `-m0x100000000:0x100000000` to test.

yzt000000, 2020-11-24:
> 
> 
> I believe it only uses the lowermost 2 GiB of physical memory, but moving the base to 0x100000000 works fine for me (either with or without an apparent bug fix I just made). I configured pk with `--with-mem-start=0x100000000` and ran spike with `-m0x100000000:0x100000000` to test.

I had configure pk  --with-mem-start=0x100000000 , i am run the bbl on fpga platform. only the ddr mapping had been changed from 0x80000000 to 0x100000000,  then cpu will enter poweroff .

aswaterman, 2020-11-24:
Given that it works on spike, it sounds like it's not a problem with pk.

yzt000000, 2020-11-24:
I found when I modify the custom.dts 's memory define will lead to fail boot linux, 
so does the bbl can't handle this type address?
1. work version:
	memory @80000000{
		#address-cells = <1>;
		#size-cells = <1>;
		reg = <0x80000000 0x10000000>;
	};
2. bad version:
	memory @80000000{
		#address-cells = <2>;
		#size-cells = <2>;
		reg = <0x0 0x80000000 0x0 0x10000000>;
	};

yzt000000, 2020-11-25:
> I found when I modify the custom.dts 's memory define will lead to fail boot linux,
> so does the bbl can't handle this type address?
>
> 1. work version:
>    memory @80000000{
>    #address-cells = <1>; [...]

in dts file， address-cell use parent node’s define

closed 2020-11-25 by yzt000000