riscv-software-src/riscv-pk #229: pk make error

muhammadsalmanafzal, 2020-12-02:
Hey there, I'm trying to compile riscv-pk with following configure commands

```
sudo CC=/home/salman/rv32/bin/riscv32-unknown-elf-gcc CFLAGS="-O3 -Wall" ./configure --prefix=/home/salman/toolchains/pk --host=riscv32-unknown-elf
```

and the output of this is:

```checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv32-unknown-elf
checking for riscv32-unknown-elf-gcc... /home/salman/rv32/bin/riscv32-unknown-elf-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether /home/salman/rv32/bin/riscv32-unknown-elf-gcc accepts -g... yes
checking for /home/salman/rv32/bin/riscv32-unknown-elf-gcc option to accept ISO C89... none needed
checking for riscv32-unknown-elf-g++... no
checking for riscv32-unknown-elf-c++... no
checking for riscv32-unknown-elf-gpp... no
checking for riscv32-unknown-elf-aCC... no
checking for riscv32-unknown-elf-CC... no
checking for riscv32-unknown-elf-cxx... no
checking for riscv32-unknown-elf-cc++... no
checking for riscv32-unknown-elf-cl.exe... no
checking for riscv32-unknown-elf-FCC... no
checking for riscv32-unknown-elf-KCC... no
checking for riscv32-unknown-elf-RCC... no
checking for riscv32-unknown-elf-xlC_r... no
checking for riscv32-unknown-elf-xlC... no
checking for g++... g++
configure: WARNING: using cross tools not prefixed with host triplet
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for riscv32-unknown-elf-ar... no
checking for ar... ar
checking for riscv32-unknown-elf-ranlib... no
checking for ranlib... ranlib
checking for riscv32-unknown-elf-readelf... no
checking for readelf... readelf
checking for riscv32-unknown-elf-objcopy... no
checking for objcopy... objcopy
checking for a BSD-compatible install... /usr/bin/install -c
checking how to run the C preprocessor... /home/salman/rv32/bin/riscv32-unknown-elf-gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
configure: configuring default subproject : pk
configure: configuring default subproject : bbl
configure: configuring default subproject : softfloat
configure: configuring default subproject : dummy_payload
configure: configuring default subproject : machine
configure: configuring default subproject : util
configure: creating ./config.status
config.status: creating pk.mk
config.status: creating bbl.mk
config.status: creating softfloat.mk
config.status: creating dummy_payload.mk
config.status: creating machine.mk
config.status: creating util.mk
config.status: creating Makefile
config.status: creating config.h
config.status: config.h is unchanged
```

and when I make it

```/home/salman/rv32/bin/riscv32-unknown-elf-gcc -Wl,--build-id=none -nostartfiles -nostdlib -static    -fno-stack-protector -o dummy_payload dummy_payload.o -L.  -ldummy_payload -lgcc -Wl,--defsym=MEM_START=0x80000000,-T,./dummy_payload/dummy_payload.lds
/home/salman/rv32/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: cannot open output file dummy_payload: Is a directory
collect2: error: ld returned 1 exit status
make: *** [Makefile:336: dummy_payload] Error 1
```

I think the problem is the configure where it is unable to find riscv-gcc like in here:

```checking for riscv32-unknown-elf-g++... no
checking for riscv32-unknown-elf-c++... no
checking for riscv32-unknown-elf-gpp... no
checking for riscv32-unknown-elf-aCC... no
checking for riscv32-unknown-elf-CC... no
checking for riscv32-unknown-elf-cxx... no
checking for riscv32-unknown-elf-cc++... no
checking for riscv32-unknown-elf-cl.exe... no
checking for riscv32-unknown-elf-FCC... no
checking for riscv32-unknown-elf-KCC... no
checking for riscv32-unknown-elf-RCC... no
checking for riscv32-unknown-elf-xlC_r... no
checking for riscv32-unknown-elf-xlC... no
```

I've tried similar procedure with riscv64-unknown-elf-gcc but same error is given. Actually, I was trying to run "spike pk <some_file>" but it was unable to find pk and hence I'm here. Can you kindly help me solve the problem? I'll appreciate it.

muhammadsalmanafzal, 2020-12-06:
I figured out this one. After **./configure** is run, all the new files are made in the same directory, so when **make** is run, it picks up **dummy_payload** directory instead of the required file. The way to solve is make a **build** directory and run your configure there to remove the clash.

closed 2020-12-06 by muhammadsalmanafzal

aswaterman, 2020-12-08:
I see, thanks for tracking it down.  Many projects don't support building in the source directory, but it would probably be preferable if configure issued an error, rather than failing in a confusing way.